mantidproject/mantid #37499: LoadNexusProcessed Is Unreliable on macOS

cailafinn, 2024-06-12:
**Describe the bug**
This defect was introduced by https://github.com/mantidproject/mantid/pull/37273
It is a intermittent error that only happens on macOS. The exact error:
```log
21:08:49 LoadNexusProcessed-[Error] Error in execution of algorithm LoadNexusProcessed:
21:08:49 LoadNexusProcessed-[Error] Attempt to load from an empty dataset /mantid_workspace_1/peaks_workspace/column_14
21:08:49 libc++abi: terminating due to uncaught exception of type std::runtime_error: Attempt to load from an empty dataset /mantid_workspace_1/peaks_workspace/column_14
21:08:49 Subprocess aborted
```

**To Reproduce**
1. Occurs occasionally on our macOS runners. Need to find a reliable way to reproduce locally. 

**Expected behavior**
Test should always pass if it passes most of the time. 

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

cailafinn, 2024-06-13:
This isn't actually an unreliable test. Running it twice in a script will crash on macOS, the test just revealed the failure. Tested on the nightly, mantid 6.9.1 and mantid 6.7. All crash. 

It only seems to happen with the `SingleCrystalLeanElasticPeakTable.nxs` file, so it's now a question of working out what's wrong with the file or wrong with the loading process for those types of files.

zjmorgan, 2024-06-13:
Interesting. Thanks for tracking it down. Yes, I added the file. The context was to preserve the peak shape when saving/loading lean peaks workspaces.

zjmorgan, 2024-06-13:
The file was created loading `SingleCrystalPeakTable.nxs` and converting to `LeanElasticPeak`

PR #37515 (2024-06-14): Update File Used in LoadNexusProcessedTest for LeanElasticPeaks Workspaces
### Description of work

#### Summary of work
There seems to be something wrong with the current file which is causing heap-breaking errors on macOS. I've created a new file from running the scripts in [the documentation](https://docs.mantidproject.org/nightly/concepts/LeanElasticPeaksWorkspace.html), and using that I don't get the same failures. This should improve the stability of the test. 

Merging into `release-next` for the sake of the nightly pipelines.

Fixes #37499

#### Further detail of work
- The test is fun [... cut]
closed 2024-06-14 by github-actions[bot]