mantidproject/mantid #37514: Instrument view crashed on peak overlay

RichardWaiteSTFC, 2024-06-13:
**Describe the bug**
Overlaying a specific peaks workspace crashes instrument view - the vast majority of peak workspaces are fine so this has been hard to reproduce. But I now have some files for which it is consistently reproducible.

**To Reproduce**
(1) Load in this workspaces (on my babylon)

The instrument workspace is
`"\\olympic\Babylon5\Public\RWaite\UBs_NickF\SXD34361_processed.nxs"`
The peaks workspace is
`"\\olympic\Babylon5\Public\RWaite\UBs_NickF\SXD34361_peaks_predicted.nxs"`

If these aren't available let me know I have a copy on my desktop

(2) Open the instrument workspace `SXD34361_processed` in instrument viewer

(3) Overlay the peaks workspace `SXD34361_peaks_predicted` by dragging from the workspace list onto the instrument viewer

**Expected behavior**
Not crash, would overlay peaks

**Platform/Version (please complete the following information):**
 - OS: seen on IDAaaS and windows
 - Mantid Version seen on v6.8 onwards (including nightly 6.9.20240607)

jclarkeSTFC, 2024-06-14:
This particular example is caused by a peak having a detector ID of -1:

[image]

Maybe the other peak workspaces also have the same thing? We can improve it so that this doesn't take down Mantid in future.

RichardWaiteSTFC, 2024-06-14:
Thanks for that - yes this can happen if you predict peaks outside detector coverage for instruments that support `extended detector space`  (e.g. WISH). For instruments that don't it makes sense not to plot them - it would be worth checking on WISH using `PredictPeaks` and predict outside detector

jhaigh0, 2024-06-17:
Did you move the file you mentioned? I can only see these for SXD34361
[image]

RichardWaiteSTFC, 2024-06-17:
I think all you need is a peak table with a null detector ID - here is the one I was using (renamed to .txt)
[SXD34361_peaks_predicted.nxs.txt](https://github.com/user-attachments/files/15866037/SXD34361_peaks_predicted.nxs.txt)

PR #37645 (2024-07-25): Draw peak marker using cached position if no detector
Peaks have a cached position, but they may not have a detector, so we can't just rely on the detector ID to draw the marker.

Fixes #37514 

Peaks have a cached position, which is where the position is stored when there's no corresponding detector. This can happen because it is possible to have peaks beyond the bounds of any detector, which is what is occurring the files in the linked issue. I added a method to the `UnwrappedSurface` base class to project a position onto the surface, so in `PeakOverlay::createMarkers()` we can use that meth [... cut]
closed 2024-07-25 by RichardWaiteSTFC