mantidproject/mantid #37521: Unpin cmake

SilkeSchomann, 2024-06-14:
Cmake was pinned to <3.27 to avoid problems with updating MSlice as an external project. 

The setting for UPDATE_DISCONNECTED gets ignored otherwise.

From cmake docs [here](https://cmake.org/cmake/help/latest/module/ExternalProject.html):

Changed in version 3.27: When UPDATE_DISCONNECTED is true, the update step will be executed if any details about the update or download step are changed. Furthermore, if using the git download/update method, the update logic will be modified to skip attempts to contact the remote. If the GIT_TAG mentions a ref that is not known locally, the update step will halt with a fatal error.

jclarkeSTFC, 2024-07-23:
How do we recreate this problem?

SilkeSchomann, 2024-07-23:
A new version of MSlice is only downloaded if it is either a new environment or the SHA in scripts/ExternalInterfaces/CMakeLists.txt was changed, otherwise the version of cmake is not relevant.
If you want to re-use the environment, you could use an MSlice SHA that is not for the head of the main branch I think.

jclarkeSTFC, 2024-07-23:
I'm using cmake 3.30.1 on Windows and it runs cmake okay when changing the mslice SHA, maybe I'll try a PR and see what happens.

SilkeSchomann, 2024-07-23:
I think the problem occurred for the first time in cmake 3.27. Then it seemed to be fixed, but came back in version 3.29. So it is entirely possible that cmake 3.30.1 is ok again. In addition, it would be easy to detect and fix if this problem comes back during the next beta testing phase.

PR #37687 (2024-07-31): Unpin cmake to allow >3.26
Previously there was a problem with the UPDATE_DISCONNECTED option in cmake.

I disabled a warning about the `FindBoost` module by setting the `cmake` policy, see [here](https://cmake.org/cmake/help/latest/module/FindBoost.html). It recommends using the `BoostConfig.cmake` config file from the Boost repository, but the version of Boost that we're currently using doesn't appear to have that file. 

The pattern that we were using for including `googletest` and `span` is now deprecated, see [here](https://cmake.org/cmake/help/latest/module/FetchContent.html#command:fetc [... cut]
closed 2024-07-31 by thomashampson