mantidproject/mantid #37547: Windows Interface Smoke Tests  :facepalm:

koshchii, 2024-06-19:
We must test both the .exe installer package and install direct from conda.

* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n mantid_rc1 -c mantid/label/v6.10.0rc1 mantidworkbench --yes
mamba activate mantid_rc1
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :facepalm:

## ISIS only, if possible, so catalogue access can be tested

- [ ] MantidWorkbench opens without errors or warnings 
- [ ] Every option in `Interface` opens a GUI 
- [ ] Load some test data 
- [ ] Access Catalogue through algorithms (use Facilities account, same as for IDAaaS): [CatalogLogin](https://docs.mantidproject.org/algorithms/CatalogLogin-v1.html) and [CatalogGetDataFiles](https://docs.mantidproject.org/algorithms/CatalogGetDataFiles-v1.html)
- [ ] Saving/loading projects works 
 - [ ] Alter preferences in [File > Settings](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/01_save_settings.html#settings) and check they are obeyed

MohamedAlmaki, 2024-06-20:
I got "DllOpen-[Error] Could not open library C:\Users\joy22959\AppData\Local\mambaforge\envs\mantid_rc1\Library\plugins\qt5\MantidScientificInterfacesIndirectQt5.dll: failed with error 126: The specified module could not be found." error 
Indirect category is missing in the interfaces menu

MohamedAlmaki, 2024-06-20:
When I try to open Muon Elemental Analysis interface I got this error:
```
Continue working.
Traceback (most recent call last):
  File "C:\Users\joy22959\AppData\Local\mambaforge\envs\mantid_rc1\lib\site-packages\workbench\app\mainwindow.py", line 392, in <lambda>
    action.triggered.connect(lambda checked_py, script=script: self.launch_custom_python_gui(script))
  File "C:\Users\joy22959\AppData\Local\mambaforge\envs\mantid_rc1\lib\site-packages\workbench\app\mainwindow.py", line 352, in launch_custom_python_gui
    self.interface_executor.execute(open(filename).read(), filename)
  File "C:\Users\joy22959\AppData\Local\mambaforge\envs\mantid_rc1\lib\site-packages\mantidqt\widgets\codeeditor\execution.py", line 153, in execute
    executor.execute(code_str, filename, flags, self.globals_ns, line_offset)
  File "C:\Users\joy22959\AppData\Local\mambaforge\envs\mantid_rc1\lib\site-packages\mantidqtinterfaces\Elemental_Analysis.py", line 8, in <module>
    from mantidqtinterfaces.Muon.GUI.ElementalAnalysis.elemental_analysis import ElementalAnalysisGui
  File "C:\Users\joy22959\AppData\Local\mambaforge\envs\mantid_rc1\lib\site-packages\mantidqtinterfaces\Muon\GUI\ElementalAnalysis\elemental_analysis.py", line 12, in <module>
    from mantidqtinterfaces.Muon.GUI.ElementalAnalysis.PeriodicTable.periodic_table_view import PeriodicTableView
  File "C:\Users\joy22959\AppData\Local\mambaforge\envs\mantid_rc1\lib\site-packages\mantidqtinterfaces\Muon\GUI\ElementalAnalysis\PeriodicTable\periodic_table_view.py", line 9, in <module>
    from mantidqtinterfaces.Muon.GUI.ElementalAnalysis.PeriodicTable import periodic_table
  File "C:\Users\joy22959\AppData\Local\mambaforge\envs\mantid_rc1\lib\site-packages\mantidqtinterfaces\Muon\GUI\ElementalAnalysis\PeriodicTable\periodic_table.py", line 268, in <module>
    _correct_with_peak_data_file()
  File "C:\Users\joy22959\AppData\Local\mambaforge\envs\mantid_rc1\lib\site-packages\mantidqtinterfaces\Muon\GUI\ElementalAnalysis\PeriodicTable\periodic_table.py", line 255, in _correct_with_peak_data_file
    with open(PeriodicTableModel().get_default_peak_data_file(), "r") as data_file:
  File "C:\Users\joy22959\AppData\Local\mambaforge\envs\mantid_rc1\lib\site-packages\mantidqtinterfaces\Muon\GUI\ElementalAnalysis\PeriodicTable\periodic_table_model.py", line 16, in __init__
    self.load_peak_data()
  File "C:\Users\joy22959\AppData\Local\mambaforge\envs\mantid_rc1\lib\site-packages\mantidqtinterfaces\Muon\GUI\ElementalAnalysis\PeriodicTable\periodic_table_model.py", line 24, in load_peak_data
    with open(self._peak_data_file, "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\joy22959\\AppData\\Local\\mambaforge\\envs\\mantid_rc1\\lib\\site-packages\\mantid\\utils\\muon\\peak_data.json'
```

rbauststfc, 2024-06-20:
For the Windows standalone installation, I also get the same error when trying to open the Muon Elemental Analysis interface. However, I don't seem to have the issue with the missing DLL, and I can see the Indirect option on the Interfaces menu.

MohamedAlmaki, 2024-06-20:
The TOF converter seems to be too compressed, making it unusable:
[image]

MohamedAlmaki, 2024-06-20:
@rbauststfc are you happy to close it if you haven't found any issue?

rbauststfc, 2024-06-20:
Yes, I think we can close this one now, thanks @MohamedAlmaki

closed 2024-06-20 by rbauststfc