mantidproject/mantid #37554: MacOS Interface Smoke Tests  :facepalm:

koshchii, 2024-06-19:
We must test both the .dmg installer package and install direct from conda.

* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n mantid_rc1 -c mantid/label/v6.10.0rc1 mantidworkbench --yes
mamba activate mantid_rc1
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :facepalm:

## ISIS only, if possible, so catalogue access can be tested

- [ ] MantidWorkbench opens without errors or warnings 
- [ ] Every option in `Interface` opens a GUI 
- [ ] Load some test data 
- [ ] Access Catalogue through algorithms (use Facilities account, same as for IDAaaS): [CatalogLogin](https://docs.mantidproject.org/algorithms/CatalogLogin-v1.html) and [CatalogGetDataFiles](https://docs.mantidproject.org/algorithms/CatalogGetDataFiles-v1.html)
- [ ] Saving/loading projects works 
 - [ ] Alter preferences in [File > Settings](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/01_save_settings.html#settings) and check they are obeyed

SilkeSchomann, 2024-06-20:
Muon/Elemental Analysis crashes Mantid when trying to open.

Garnet, SNAPRed and Shiver show warning that they are not available.

SilkeSchomann, 2024-06-20:
I've raised an issue for the problem with Elemental Analysis: https://github.com/mantidproject/mantid/issues/37582

SilkeSchomann, 2024-06-20:
Garnet, SNAPRed and Shiver are not included into Mantid by default, so this behaviour is as expected.

closed 2024-06-20 by SilkeSchomann