mantidproject/mantid #37571: IDAaaS Workspace/Algorithm Smoke Tests   :man_astronaut: :woman_astronaut: :space_invader:

koshchii, 2024-06-19:
An IDAaaS session link will be shared with you. It will either be a no-GPU or a GPU IDAaaS instance. All IDAaaS-installed packages are Conda installs.

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :man_astronaut: :woman_astronaut: :space_invader:

* Load some workspaces, try 5 from different instruments
* Show detectors on these workspaces
* Check Instrument Viewer for these workspaces, follow  documentation for  [Instrument Viewer Widget](https://docs.mantidproject.org/workbench/instrumentviewer.html) and some more detail on the picking tools in the [Basic Course](https://docs.mantidproject.org/tutorials/mantid_basic_course/connecting_data_to_instruments/03_investigating_data.html)
  - [ ] Render tab 
  - [ ] Draw tab 
  - [ ] Pick tab 
  - [ ] Instrument tree tab 
* Test 5 - 10 random algorithms, follow [how to run Algorithms](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/algorithms_workspaces_and_histories/02_algorithms.html#algorithms) and [pick your favourite](https://docs.mantidproject.org/algorithms/)
 - [ ] Help button 
 - [ ] Validation of inputs 
 - [ ] Overwriting workspace 
 - [ ] Run the algorithm 
- [ ] Check right-click options of the workspaces, they are listed [in the image in the documentation](https://docs.mantidproject.org/nightly/workbench/workspacetreewidget.html#workbenchworkspacetoolbox).

warunawickramasingha, 2024-06-20:
Raised #37583 but this is **not** a regression issue.

warunawickramasingha, 2024-06-20:
Followed the given steps and no other issues were observed.

closed 2024-06-21 by adriazalvarez