mantidproject/mantid #37579: User defined fit function box hidden behind scroll bar

jclarkeSTFC, 2024-06-20:
**Describe the bug**

If there is a scroll bar in the fit options, then it's impossible to click the button in the `Formula` box in order to open the User Function Dialog.

**To Reproduce**
Open the fit toolbar on a plot, size the window vertically such that a scroll bar appears, add a `UserFunction`, then try and open the User Function Dialog

**Expected behavior**

Button can be clicked

**Screenshots**

[image]

**Platform/Version (please complete the following information):**
 - OS: macOS
 - OS Version: 12.7.5
 - Mantid Version 6.10.0rc1

**Additional context**

Found during smoke testing.

jhaigh0, 2024-07-01:
Possibly mac specific, looks okay on windows

[image]

jhaigh0, 2024-07-17:
Had a look on ubuntu and looks okay, so just a mac thing

[image]

SilkeSchomann, 2025-02-04:
I've just re-tested this on macOS with Mantid nightly from 17th of January 2025. It is not possible anymore to resize the window vertically so that a scrollbar appears. As a consequence, the `User Function Dialog` button always remains accessible.

SilkeSchomann, 2025-02-24:
This issue appeared again during smoke testing and seems to affect at least some installations on macOS>

closed 2025-04-03 by thomashampson