mantidproject/mantid #37584: TOF Convertor menu appears compressed in high resolution machines

MohamedAlmaki, 2024-06-20:
**Describe the bug**

When opening the TOF converter interface, it appears too minimized, making it unusable.  
**To Reproduce**

You need a high resolution machine to replication the issue
1. Go to interfaces->utility->TOF convertor
2. The window will appear small and unusable

**Expected behavior**

It should appear in normal size with adjustable dimensions
**Screenshots**

[image]

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS] Windows 
 - OS Version: 10
 - Mantid Version [e.g. 6.0.0] 6.10.rc1

jclarkeSTFC, 2024-06-21:
Is this the same in 6.9.1?

MohamedAlmaki, 2024-06-24:
Yes, it is not a regression bug

PR #37666 (2024-07-17): Fix TOF converter on high res screens
### Description of work

Removed the minimum size from the tof converter ui file to fix it being "squished" on hi-res screens.

Fixes #37584 

### To test:

see original issue for what it used to look like.

- I have tested on a 4k monitor, if you have one, open the interface and check it looks okay upon opening.
- If you don't have one, check it looks okay on a standard hd monitor.

---

### Reviewer

Please comment on the points listed below ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)).
**Your comments will be used  [... cut]
closed 2024-07-17 by SilkeSchomann