mantidproject/mantid #37585: 3D Surface plot zooming in and out with mouse hover

GuiMacielPereira, 2024-06-20:
**Describe the bug**
3D surface plots incorrectly trigger zoom when the log axis scale of the colorbar is selected.

**To Reproduce**

1. Load LOQ74041
2. Click `Plot > 3D > Surface`
3. Right-click on the plot > Color bar 
4. That shoud be enough to trigger the bug
5. If you hover the mouse over the plot you'll notice it zooms in and out
6. You can stop it if you click on the plot
7. The scroll wheel doesn't seem to work.

**Expected behavior**
Fix the zooming with hovering and maybe get the scroll wheel working for the zoom.

**Screenshots**
Zoomed out plot:
[image]

**Platform/Version (please complete the following information):**
 - OS: Ubuntu
 - OS Version: 22.04
 - Mantid Version 6.10

**Additional context**

jclarkeSTFC, 2024-06-21:
I can't get it to do this on IDAaaS, I'll have a go on my Ubuntu VM

jhaigh0, 2024-07-15:
I can get this to happen on windows 👍

PR #37660 (2024-07-18): Fix 3d plot bug where zoom was toggled by context menu
### Description of work

By default, holding right click and panning the mouse around on a Axes3D plot with zoom in and out.
In workbench, right-clicking a plot with bring up a context menu (for 3d plots its changing the colour bar scale).
Opening this QMenu seems to eat the mouse button release signal, so after you've used the menu the plot things right click is still held down.

I tried to fix the core problem of the release signal never getting caught but struggled to find a solution. Instead, I have just unassigned the zoom button on M [... cut]
closed 2024-07-18 by thomashampson