mantidproject/mantid #37627: Consolidate dependency requirements

thomashampson, 2024-07-02:
**Describe the outcome that is desired.**
Dependencies and version pinning requirements are currently spread across several files: the three `mantid-developer` environment files and the conda-build recipes. This duplication can easily lead to version pinnings becoming out of sync between developer environments and the conda-build recipes, meaning that we are testing against versions of dependencies that are different to the ones we are shipping against.

There are two separate but related outcomes that would be nice to resolve by consolidating requirements:
- Specify our dependency versions in one place to be shared between the developer and conda-build environments.
- Have a single mantid-developer environment definition to be shared between all OS. Currently there is no support for specifying that a particular package is only installed on a single OS (see [here](https://github.com/conda/conda/issues/8089) for a discussion on that).

**Describe any solutions you are considering**
1.  Create a `mantid-developer` [conda metapackage](https://docs.conda.io/projects/conda-build/en/latest/resources/commands/conda-metapackage.html) to be uploaded to the `mantid` channel.
Advantages:
  - can easily use the same pinnings as the other package recipes from `conda_build_config.yaml`
  - developers can easily create multiple development environments for testing changes to dependencies without having to rename the environment name in the current mantid-developer file.
Disadvantages:
  - would require an additional stage in the nightly pipeline, and at the start of all PR checks to re-build the mantid-developer conda package. In the nightly pipeline, we would deploy the package. In PR checks we would first check whether there are any changes to the recipe, and then build and install a local metapackage.
  - new metapackage would only be available after it has been deployed by the nightly pipeline, and wouldn't be directly in sync with `main`
  - might cause issues if there are differences between developer environments between `main` and `release-next`

2.  Write a python script to parse a single yaml file cantaining OS selectors to generate an environment file and inject the version constraints from `conda_build_config.yaml`. This is the approach that was taken by the mantid imaging team (see [here](https://github.com/mantidproject/mantidimaging/blob/1cd9d21191eef1df5ff0deec0b0a0db4272397f4/setup.py#L83)), where you run this command to create the developer environment:
`python3 ./setup.py create_dev_env`
Advantages:
  - developer env is always configured **directly** from source
  - does not require potentially lengthy conda-build step
  - no need to host a metapackage on the mantid channel
Disadvantages:
  - it's not a standard way of creating a conda environment
  - may be introducing new bugs

**Additional context**
This is essentially the same as this issue: https://github.com/mantidproject/mantid/issues/33442, but since the conda recipes have moved into the mantid repo, a lot of the information there is out of date.

samtygier-stfc, 2024-07-11:
Some comments on why we chose a python script to do this in Mantid Imaging:

- Allows specifying dependency version in just a single place
- Allows OS specific dependencies (environment.yml does not support this https://github.com/conda/conda/issues/8089)
- Avoids chicken and egg issues around needing a conda package to exist in order get the dependencies in order to build a package. This was a problem when ever we did a PR to update packages

https://github.com/mantidproject/mantidimaging/pull/861
https://github.com/mantidproject/mantidimaging/issues/998

samtygier-stfc, 2024-07-11:
There is also some useful discussion in https://github.com/conda/conda/issues/10633

peterfpeterson, 2024-08-20:
Link to [unidep](https://unidep.readthedocs.io/en/latest/index.html) which would support option 2

PR #37924 (2024-09-12): mantid-developer metapackage
### Description of work
Adds a new Conda recipe that generates a `mantid-developer` Conda [metapackage](https://docs.conda.io/projects/conda/en/latest/user-guide/concepts/packages.html#metapackages). The metapackage is a replacement for the `mantid-developer-{linux,osx,win}.yml` environment files that were used to create development environments. The major advantage of this is that dependency version numbers only appear in one place. There were still a few instances where the version pinning was different in the developer and packaging environments. Those should al [... cut]
closed 2024-09-12 by SilkeSchomann