mantidproject/mantid #37639: Crash when loading a workspace group into the Bayes Fitting interface

rbauststfc, 2024-07-05:
**Original reporter:** Mona Sarter at ISIS

**Describe the bug**

The Bayes Fitting interface crashes when trying to load a workspace group from file, or as a workspace, on the Quasi tab.

**To Reproduce**

Note, Mona has provided a specific file that fails. This file loads as a workspace group, so I suspect (but haven't tested) that this could be replicated with any file that loads as a workspace group.

1) Go to `Interfaces -> Inelastic -> Bayes Fitting`.
1) Click on the Quasi tab.
1) Browse to load the file. Wait for a while as it loads. After a small delay, Mantid will crash. The source appears to be in Quasi.cpp in `handleSampleInputReady`, where we retrieve the workspace from the ADS and cast to a `MatrixWorkspace` but don't do a `nullptr` check.

You can also replicate as follows:

1) Load the file, which should load as a workspace group.
1) Go to `Interfaces -> Inelastic -> Bayes Fitting`.
1) Click on the Quasi tab.
1) Change the dropdown next to the Sample input from `File` to `Workspace`.
1) From the dropdown, select the workspace group name (it's right at the bottom of the list), and Mantid will crash. I haven't debugged the source of this, but it may be the same as above.

**Expected behavior**

Mantid should not crash and should instead handle this in some appropriate way.

**Platform/Version (please complete the following information):**
 - OS: originally encountered on Windows, replicated on IDAaaS
 - Mantid Version: found in 6.9.1 and replicated in current Nightly. Can only replicate the first of the two set of steps in 6.8, as the second doesn't seem to populate the workspace dropdown on IDAaaS.

**Additional context**

I haven't checked if we get the same behaviour for the other tabs and inputs in the Bayes Fitting interface, so probably worth doing a thorough check when working on this issue.

SpencerHowells, 2024-07-05:
@robertapplin Using wrong format of data. They do not use the correct data reduction program. It's creating an nxspe file (old format) instead of nxs (new format) so you cannot run Sqw. I sorted that out with Le Duc but users still use the old!

SpencerHowells, 2024-07-05:
@robertapplin How about just checking if the ws is nxspe. If so, stop and give instructions that the data should be reduced to an nxs file and converted to Sqw using the interface.

PR #37642 (2024-07-19): Prevent crash on Bayes Fitting Quasi
**Report to:** Mona Sarter and Spencer Howells

### Description of work
This PR fixes several issues on the Quasi tab of the Inelastic Bayes Fitting interface:
- The Sample and Resolution Data Selectors and Run button are now disabled while data is loading
- If a dataset does not have an "analyser" or "resolution", then a check to see if these two match for the sample and resolution is now skipped.  This is because we want to be able to use this interface for LET data, which do not have an equivalent for a instrument "analyser" or "resolution"
- The data se [... cut]
closed 2024-07-19 by peterfpeterson