mantidproject/mantid #37654: run_number in nxspe files

abuts, 2024-07-12:
**Is your feature request related to a problem? Please describe.**

Horace package uses nxspe files produced by Mantid as its primary source of data about neutron scattering. Currently we are enabling the possibility to combine information from different experients and differnet instruments so the information in nxspe files should be easy identifiable.  It currently contains information about instrument (unique insturment name) while information about experiment is not there and should be extracted using various different means. 

Facilities have long developed unique vay of identifying experiments and this way is run numbers, so these numbers can be used for identifying experiments in Horace too. 

**Describe the solution you'd like**

Alongside with instrument name, I want to add run_number field, which would contain the run number taken from processed workspace. 
E.g. 
Current state of nxspe file:

    ws_name->instrument->name

Suggested change:

    ws_name->instrument->name
    ws_name->instrument->run_number

**Additional context**
Additional hdf field should not affect any software not interested in it (so Mantid loader should not be changed unless Mantid wants to recover this information.

It looks like the nxspe format version from
 
  ws_name->definition(attribute)->version

should be also changed from 1.2 to 1.3 but this is optional and can be rejected if causes any problem

SpencerHowells, 2024-07-12:
@abuts Could you also add a line to set efixed=Ei so that several of our programs can run without having to do it manually, thanks.

abuts, 2024-07-12:
you mean duplicate `fixed_energy` field from

   ws_name->NXSPE_info->fixed_energy

with additional field:

   ws_name->NXSPE_info->efixed

Do not see problem with this though SNS was not very happy with such changes in a past. But if there are no objections from the community -- why not.

SpencerHowells, 2024-07-12:
@abuts We have to use SetInstrumentParameter so it's an instrument parameter together with emode whcih should be Direct.

abuts, 2024-07-12:
@SpencerHowells  -- I do not understand. SaveNXSPE currently saves EI for direct spectrometers as field 
ws_name->NXSPE_info->fixed_energy

SpencerHowells, 2024-07-12:
@abuts According to the SaveNXSPE info there is an input Efixed which is optoinal - this can br Ei. All the Direct instruments we use for QENS use efixed. At ISIS the script we use to ceate Sqw has option of NXSPE or NXS for oiutput and Duc Le now recommends we always use NXS. ILL and Munich now also use efixed.

abuts, 2024-07-12:
ok, so efix on reduced workspace for indirect instruments is named differently and saveNXSPE does not pick it up to save as `fixed_energy`. Makes sence to fix it indeed. 

I am sure nobody would object this.

mducle, 2024-07-12:
@SpencerHowells There is already a `fixed_energy` field in the [NXSPE specification](https://hdf5.gitlab-pages.esrf.fr/nexus/nxdata_transformation/classes/applications/NXspe.html). I think the main problem is that some Mantid algorithms don't save this information, but rather create a log entry with the Efix value instead - and the log is discarded when saving to `.nxspe` because logs are not in the NXSPE specs.

In any case we would recommend to use `.nxs` (or `_red.nxs`) to QENS users instead of `.nxspe` and as far as I know the regular reduction script will generate `_red.nxs` files compatible with the QENS data analysis GUIs and we just need to update the polarised reduction files used on LET.

PR #37673 (2024-09-10): Add run number to output of SaveNXSPE
### Description of work

The run number of the input workspace is now added to the instrument section .nxspe files. 

I have also updated the nxspe file version number.

Fixes #37654.

**Report to:** @abuts 

### To test:

So far, I have ensured that an .nxspe file saved with the updated version of SaveNXSPE can still be opened with DAVE - MSlice. 

To verify that the run number is set as expected, I have used [HDFView](https://www.hdfgroup.org/downloads/hdfview/).

It would be good to check if Horace can use the run number written to the .nxspe file.

fro [... cut]
closed 2024-09-10 by robertapplin