mantidproject/mantid #37720: Fix PowderReduceP2DTest to run on Windows and OSX

sf1919, 2024-07-31:
**Describe the bug**
The nightly has been failing for a week on OSX builds because of this test. Looking at the code it seems it was disabled on Windows in January 2023. It is not acceptable to turn this off on OSX too now and instead should be fixed.

Relevant console logs
- https://builds.mantidproject.org/view/Nightly%20Pipelines/job/main_nightly_deployment/847/execution/node/221/log/?consoleFull
- https://builds.mantidproject.org/view/Nightly%20Pipelines/job/main_nightly_deployment/845/execution/node/180/log/?consoleFull

**To Reproduce**
- Run test on Windows 
- Run test on ARM Mac

**Expected behavior**
Test should pass on all operating systems.

**Platform/Version (please complete the following information):**
 - OS: Windows and ARM MacOS
 - Mantid Version : Nightly from 25th July onwards

**Additional context**
The system test does run on older MacOS machines but for some reason fails on our M1 Macs in the pipeline, even though we are cross-compiling back to x86. We think there is a possibility the failures is related to upgrading clang recently (#37573 ) but we cannot be certain. Regardless this test seems to have been causing issues on Windows for some time and so needs dealing with.

sf1919, 2024-08-15:
This appears to be working on Windows now (see #37727) so it is just OSX that it needs to be fixed on.

robertapplin, 2024-08-30:
This test is now failing again on the Windows nightly https://builds.mantidproject.org/view/Nightly%20Pipelines/job/main_nightly_deployment/867/execution/node/216/log/?consoleFull

peterfpeterson, 2024-08-30:
My suspicion is that the difference between platforms comes from `PowderReduceP2D` calling `StripVanadiumPeaks2` calling `StripPeaks` calling `FindPeaks` which does some Fourier-based algorithm followed by peak fitting.

PR #39074 (2025-03-26): Restore PowderReduceP2DTest
### Description of work
The issue reported in https://github.com/mantidproject/mantid/issues/37720 is blocking mantid nightly builds. The solution is to increase the tolerances for the compare and/or update the reference results.

#### Summary of work

Overrides `PowderReduceP2DTest` validation using `numpy` to compare test result values with tolerance.

Fixes #37720
. 

[7050: [mantid] Fix PowderReduceP2DTest to run on Windows and OSX](https://ornlrse.clm.ibmcloud.com/ccm/resource/itemName/com.ibm.team.workitem.WorkItem/7050)

#### Further detail of work

* ~~https [... cut]