mantidproject/mantid #37755: LoadDspacemap doctest writes ~700,000 lines of output into the doctest log

thomashampson, 2024-08-06:
**Describe the bug**
The `algorithms/LoadDspacemap-v1` doc causes ~700,000 lines of output in the doctest logs, making it difficult to read. Here is a snippet:
```
LoadDspacemap-[Error] Detector ID = 0  Inter-Module ID = 1248
LoadDspacemap-[Notice] Misses Detector ID = 0
LoadDspacemap-[Error] Detector ID = 1  Inter-Module ID = 1248
LoadDspacemap-[Notice] Misses Detector ID = 1
LoadDspacemap-[Error] Detector ID = 2  Inter-Module ID = 1248
LoadDspacemap-[Notice] Misses Detector ID = 2
LoadDspacemap-[Error] Detector ID = 3  Inter-Module ID = 1248
LoadDspacemap-[Notice] Misses Detector ID = 3
LoadDspacemap-[Error] Detector ID = 4  Inter-Module ID = 1248
LoadDspacemap-[Notice] Misses Detector ID = 4
```

**To Reproduce**
Run the doctests.

**Expected behavior**
Not all of those lines of error.

sf1919, 2024-08-15:
I'm assinging this to the ORNL Team as the algorithm is SNS specific. For some reason the file is hitting these two lines in `LoadDspacemap.cpp`
https://github.com/mantidproject/mantid/blob/main/Framework/DataHandling/src/LoadDspacemap.cpp#L274
https://github.com/mantidproject/mantid/blob/main/Framework/DataHandling/src/LoadDspacemap.cpp#L279

Either a better file needs to be provided as part of the usage data for the documentation test example or the warnings need to be reviewed in the algorithm.

I think @rosswhitfield was involved in some of the work on this.

rosswhitfield, 2024-08-18:
It was ORNL but not me. @peterfpeterson can find someone to work on it.

peterfpeterson, 2024-08-19:
This is a *really* old algorithm that appears to have been written back when we were using svn/trac by Janik Zikovsky. The first [page of the history](https://github.com/mantidproject/mantid/commits/main/Framework/DataHandling/src/LoadDspacemap.cpp) still includes removing the `Mantid/` directory that happened in 2015. Since then, it looks like all of the changes have been formatting and other tooling automatically changing the code. I'm going to contact the instrument team to see if we can deprecate this one. I wonder why this logging issue hasn't shown up more recently since the file was last changed 3 years ago by David Fair who updated the namespaces.

PR #37822 (2024-08-20): Change how instrument is specified to fix the doctest
Update the doctest for LoadDspacemap to use a much older VULCAN instrument definition which has all of the detector-ids in the calibration file.

Fixes #37755.

### To test:

Copy-and-paste the old and new doctest into mantidworkbench and see how small the logs are now.

*This does not require release notes* because it is updating a doctest to make the algorithm log less.

---

### Reviewer

Please comment on the points listed below ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)).
**Your comments will be  [... cut]