mantidproject/mantid #37760: Investigate APITest_AsynchronousTest

sf1919, 2024-08-07:
**Describe the outcome that is desired.**
This has proved an unreliable test recently on both PRs and the nightly pipeline. The desired outcome is to review the test to see if it can be improved to make it more stable.

**Describe any solutions you are considering**
Understand what the test is doing and why it is failing on asserts. 

**Additional context**
Relevant console log from Linux nightly build
```
00:38:01  1645/2723 Test  #490: APITest_AsynchronousTest .....................................................................................................................***Failed    4.92 sec
00:38:01  FrameworkManager-[Notice] Welcome to Mantid 6.10.20240724.1558.dev135
00:38:01  FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
00:38:01  Running 6 testsAsyncAlgorithm-[Notice] AsyncAlgorithm started
00:38:01  AsyncAlgorithm-[Notice] AsyncAlgorithm successful, Duration 0.02 seconds
00:38:01  
00:38:01  16AsynchronousTest
00:38:01  In AsynchronousTest::testExecution:
00:38:01  /jenkins_workdir/workspace/main_nightly_deployment/source/Framework/API/test/AsynchronousTest.h:101: Error: Assertion failed: !result.available()
00:38:01  AsyncAlgorithm-[Notice] AsyncAlgorithm started
00:38:01  AsyncAlgorithm-[Warning] AsyncAlgorithm: Execution cancelled by user.
00:38:01  .AsyncAlgorithmThrows-[Notice] AsyncAlgorithmThrows started
00:38:01  AsyncAlgorithmThrows-[Error] Error in execution of algorithm AsyncAlgorithmThrows:
00:38:01  AsyncAlgorithmThrows-[Error] Exception thrown
00:38:01  .AsyncAlgorithm-[Notice] AsyncAlgorithm started
00:38:01  AsyncAlgorithm-[Notice] AsyncAlgorithm started (child)
00:38:01  AsyncAlgorithm-[Notice] AsyncAlgorithm successful, Duration 0.01 seconds
00:38:01  AsyncAlgorithm-[Notice] AsyncAlgorithm started (child)
00:38:01  AsyncAlgorithm-[Notice] AsyncAlgorithm successful, Duration 0.01 seconds
00:38:01  AsyncAlgorithm-[Notice] AsyncAlgorithm successful, Duration 0.02 seconds. Processed as a workspace group
00:38:01  .AsyncAlgorithm-[Notice] AsyncAlgorithm started
00:38:01  AsyncAlgorithm-[Notice] AsyncAlgorithm started (child)
00:38:01  AsyncAlgorithm-[Warning] AsyncAlgorithm: Execution cancelled by user.
00:38:01  .AsyncAlgorithmThrows-[Notice] AsyncAlgorithmThrows started
00:38:01  AsyncAlgorithmThrows-[Notice] AsyncAlgorithmThrows started (child)
00:38:01  AsyncAlgorithmThrows-[Error] Error in execution of algorithm AsyncAlgorithmThrows:
00:38:01  AsyncAlgorithmThrows-[Error] Exception thrown
00:38:01  .
00:38:01  Failed 1 of 6 tests
00:38:01  Success rate: 83%
00:38:01  
```

jclarkeSTFC, 2024-09-27:
It would be good to see more examples of this happening, because I can't reproduce it on my local Linux

peterfpeterson, 2025-01-21:
Maybe this can be made a little easier to explore using `ctest --repeat-until-fail <n> ....`

closed 2025-01-30 by thomashampson