mantidproject/mantid #37768: Add correct error calculation to PolarizationEfficienciesWildes

rbauststfc, 2024-08-08:
The `PolarizationEfficienciesWildes` algorithm currently uses default Mantid error calculation, which is expected to over-estimate the errors. We need to add bespoke error calculations for the equations being calculated in this algorithm to improve the accuracy. The errors should be calculated using the same approach as we have done for the SANS efficiency algorithms.

This is part of the Polarised Reflectivity epic and a follow-on from issue #35682.

We will need to confirm the error equations to be implemented here. Either the scientists will need to provide these, or we can attempt to derive them using the same approach as we used for SANS and then confirm with our scientists that they are correct before implementing.

rbauststfc, 2024-09-17:
**_TBC_**

Fp:

$$
\huge f_p= \frac{I_1^{00} - I_1^{01} - I_1^{10} + I_1^{11}}{2(I_1^{00} - I_1^{01})}
$$

Error calculation:

$$
\huge \sigma_{f_p} = \sqrt{|\frac{\delta f_p}{\delta I_1^{00}}|^2 * \sigma^2_{I_1^{00}} + |\frac{\delta f_p}{\delta I_1^{01}}|^2 * \sigma^2_{I_1^{01}} + |\frac{\delta f_p}{\delta I_1^{10}}|^2 * \sigma^2_{I_1^{10}} + |\frac{\delta f_p}{\delta I_1^{11}}|^2 * \sigma^2_{I_1^{11}}}
$$

rbauststfc, 2024-09-17:
**_TBC_**

Fa:

$$
\huge f_a= \frac{I_1^{00} - I_1^{01} - I_1^{10} + I_1^{11}}{2(I_1^{00} - I_1^{10})}
$$

Error calculation:

$$
\huge \sigma_{f_a} = \sqrt{|\frac{\delta f_a}{\delta I_1^{00}}|^2 * \sigma^2_{I_1^{00}} + |\frac{\delta f_a}{\delta I_1^{01}}|^2 * \sigma^2_{I_1^{01}} + |\frac{\delta f_a}{\delta I_1^{10}}|^2 * \sigma^2_{I_1^{10}} + |\frac{\delta f_a}{\delta I_1^{11}}|^2 * \sigma^2_{I_1^{11}}}
$$

rbauststfc, 2024-09-17:
**_TBC_**

Polarizer efficiency

If calculating from mag workspace:

$$
\huge p= \frac{\phi}{2}\Bigg(\frac{(1-2f_a)I^{00}_2 + (2f_a-1)I^{10}_2 - I^{01}_2 + I^{11}_2}{(1-2f_p)I^{00}_2 + (2f_p-1)I^{01}_2 - I^{10}_2 + I^{11}_2}\Bigg) + \frac{1}{2}
$$

Error calculation:

$$
\sigma_p = \sqrt{|\frac{\delta p}{\delta I_2^{00}}|^2 * \sigma^2_{I_2^{00}} + |\frac{\delta p}{\delta I_2^{01}}|^2 * \sigma^2_{I_2^{01}} + |\frac{\delta p}{\delta I_2^{10}}|^2 * \sigma^2_{I_2^{10}} + |\frac{\delta p}{\delta I_2^{11}}|^2 * \sigma^2_{I_2^{11}} +  |\frac{\delta p}{\delta f_a}|^2 * \sigma^2_{f_a} +  |\frac{\delta p}{\delta f_p}|^2 * \sigma^2_{f_p} +  |\frac{\delta p}{\delta \phi}|^2 * \sigma^2_\phi}
$$

Otherwise, if calculating from the analyser efficiency:

$$
\huge p= \frac{\phi}{2(2a-1)} + \frac{1}{2}
$$

Error calculation:

$$
\huge \sigma_{p} = \sqrt{|\frac{\delta p}{\delta \phi}|^2 * \sigma^2_\phi + |\frac{\delta p}{\delta a}|^2 * \sigma^2_a}
$$

rbauststfc, 2024-09-17:
**_TBC_**

Analyser efficiency:

$$
\huge a= \frac{\phi}{2(2p-1)} + \frac{1}{2}
$$

Error calculation:

$$
\huge \sigma_{a} = \sqrt{|\frac{\delta a}{\delta \phi}|^2 * \sigma^2_\phi + |\frac{\delta a}{\delta p}|^2 * \sigma^2_p}
$$

rbauststfc, 2024-09-17:
**_TBC_**

### Rho:

$$
\huge \rho= 2f_p - 1
$$

Error calculation:

$$
\huge \sigma_\rho = \sqrt{|\frac{\delta \rho}{\delta f_p}|^2 * \sigma^2_{f_p}}
$$

### Alpha:

$$
\huge \alpha= 2f_a - 1
$$

Error calculation:

$$
\huge \sigma_\alpha = \sqrt{|\frac{\delta \alpha}{\delta f_a}|^2 * \sigma^2_{f_a}}
$$

### Two p minus 1:

$$
\huge tpmo= 2p - 1
$$

Error calculation:

$$
\huge \sigma_{tpmo} = \sqrt{|\frac{\delta tpmo}{\delta p}|^2 * \sigma^2_p}
$$

### Two a minus 1:

$$
\huge tamo= 2a - 1
$$

Error calculation:

$$
\huge \sigma_{tamo} = \sqrt{|\frac{\delta tamo}{\delta a}|^2 * \sigma^2_a}
$$

acaruana2009, 2024-10-29:
Having a very quick look, I think for the $\phi$, $f_p$ and $f_a$ quantities, the assumption that they are made up of independent (uncorrelated) measurements is a fair assumption. 

However, I think for the other quantities, because they use derived quantities (i.e. $f_p$, $f_a$ and $\phi$) to calculate them, I am not sure we can say all quantities are independent. It may be we have to substitute in for $f_p$, $f_a$ and $\phi$ so that all calculations are in terms of counts - i.e. in terms of $I^{xx}_{y}$, where $xx$ is the flipper state and $y$ is the type of sample used in the measurement.

Now it may be these give equivalent results for the two ways of calculating (I must admit I need to do more reading on this), but it is worth a check.

rbauststfc, 2024-10-29:
Thanks for the comments so far @acaruana2009, I get what you're saying about the derived quantities. I'll leave that one with you (definitely beyond my level of knowledge on this!) and just let me know what you find out.

RichardWaiteSTFC, 2024-12-03:
> **_TBC_**
> 
> Polarizer efficiency
> 
> If calculating from mag workspace:
> 
> p = ϕ 2 ( ( 1 − 2 f a ) I 2 00 + ( 2 f a − 1 ) I 2 10 − I 2 01 + I 2 11 ( 1 − 2 f p ) I 2 00 + ( 2 f p − 1 ) I 2 01 − I 2 10 + I 2 11 ) + 1 2
> 
> Error calculation:
> 
> σ p = | δ p δ I 2 00 | 2 ∗ σ I 2 00 2 + | δ p δ I 2 01 | 2 ∗ σ I 2 01 2 + | δ p δ I 2 10 | 2 ∗ σ I 2 10 2 + | δ p δ I 2 11 | 2 ∗ σ I 2 11 2 + | δ p δ f a | 2 ∗ σ f a 2 + | δ p δ f p | 2 ∗ σ f p 2 + | δ p δ ϕ | 2 ∗ σ ϕ 2

I think here you need to substitute in the equation for $f_a$ into $p$ then write the error in terms of partial derivatives wrt $I_{ij}$ - wolfram alpha may be required!

PR #39300 (2025-05-22): Correct error propagation on PolarizationEfficienciesWildes algorithm
### Description of work

#### Summary of work
This PR corrects the error propagation for the `PolarizationEfficienciesWildes` algorithm.

As part of this work a class `ErrorPropagation` has been added to `PolarizationCorrectionsHelpers.h`. This generic class utilizes the eigen `AutoDiff` module to automatically calculate the partials for different input variables. If there is wider interest in this class I will move it into the `kernel` in a separate PR.

The `PolarizationEfficienciesWildes` algorithm has been refactored sli [... cut]
commit c3192871 (2025-05-22): Correct error propagation on PolarizationEfficienciesWildes algorithm (#39300)

### Description of work

#### Summary of work
This PR corrects the error propagation for the
`PolarizationEfficienciesWildes` algorithm.

As part of this work a class `ErrorPropagation` has been added to
`PolarizationCorrectionsHelpers.h`. This generic class utilizes the
eigen `AutoDiff` module to automatically calculate the partials for
different input variables. If there is wider interest in this class I
will move it into the `kernel` in a separate PR.

The `PolarizationEfficienciesWildes` algorithm has been refa [... cut]