mantidproject/mantid #37840: Migrate to Numpy v2.0

peterfpeterson, 2024-08-20:
Maintenance for moving to numpy v2.0 (need issue links) ([migration guide](https://numpy.org/doc/stable/numpy_2_0_migration_guide.html))

- Need prototype(s) for how extensive the changes might be
- Will be done on a long-running branch to integrate the changes. `main` will be auto-merged into the branch
- numpy v2 supports python 3.9-3.12 ([python supported versions](https://devguide.python.org/versions/))
- boost v1.84 supports newer numpy [boost::python docs](https://www.boost.org/doc/libs/1_84_0/libs/python/doc/html/numpy/index.html)

perenon, 2024-08-28:
I just try to have a quick look on this.

First complication: conda dependencies. Mantid depends on quasielasticbayes which depends on numpy<2.0.0.

@robertapplin, I guess you are the one I should ask if the migration of quasielasticbayes to numpy 2 is feasible (?)

perenon, 2024-08-28:
Second complication: matplotlib. The compatibility between matplotlib qnd Numpy must be investigated. [This page](https://matplotlib.org/devdocs/devel/min_dep_policy.html) does not mention Numpy v2.

Even with the last penultimate version of matplotlib (3.9.1):

```
>>> import matplotlib

A module that was compiled using NumPy 1.x cannot be run in
NumPy 2.1.0 as it may crash. To support both 1.x and 2.x
versions of NumPy, modules must be compiled with NumPy 2.0.
Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

If you are a user of the module, the easiest solution will be to
downgrade to 'numpy<2' or try to upgrade the affected module.
We expect that some modules will need time to support NumPy 2.

AttributeError: _ARRAY_API not found
```

perenon, 2024-08-28:
The good news is that source code compiles with Numpy2. I noted some tests that would require a lifting, but the effort on the source code should be limited, I hope. Moreover, the migration guide is clear.

I will try to perform some tests, and keep track in this issue.

robertapplin, 2024-08-29:
> @robertapplin, I guess you are the one I should ask if the migration of quasielasticbayes to numpy 2 is feasible (?)

Yes I can look into this during the first sprint of v6.12

peterfpeterson, 2024-08-29:
Do we have confirmation as to whether matplotlib supports numpy v2 yet?

perenon, 2024-08-29:
According to [here](https://github.com/numpy/numpy/issues/26191), matplotlib 3.8.4 might be compatible. I will check

peterfpeterson, 2024-08-29:
https://github.com/matplotlib/matplotlib/issues/26778 suggests it too

perenon, 2024-08-29:
Indeed. Test feedback: the minimal matplotlib version you must install with numpy 2 is 3.8.4.

With matplotlib 3.8.4, Mantid workbench can be launched. Slice viewer opens. I did not push the test further.

With matplotlib 3.9.X, Mantid workbench does not even start.

I will try with automatic tests, but it will take a larger amount of time :-).

AndreiSavici, 2024-08-29:
I've tested with a simple environment with numpy 2.1.0, matplotlib 3.9.2, python 3.12.5

perenon, 2024-08-30:
> Indeed. Test feedback: the minimal matplotlib version you must install with numpy 2 is 3.8.4.
> 
> With matplotlib 3.8.4, Mantid workbench can be launched. Slice viewer opens. I did not push the test further. (edit: I had to remove quasielasticbayes from Mantid dependencies)
> 
> With matplotlib 3.9.X, Mantid workbench does not even start.
> 
> I will try with automatic tests, but it will take a larger amount of time :-).

For UnitTests: With matplotlib 3.8.4, Numpy 2.1.0, python 3.10.14, and after changing `np.NaN` to `np.nan`, `np.Inf` to `np.inf`, `np.mat` to `np.asmatrix` and `np.alltrue` to `np.all`

=> I am left with 21 tests failing. 9 are GUI-related and 12 are python-related (One comes from a PyStoG -a `numpy.alltrue` statement-)

perenon, 2024-08-30:
So my two cents, if we want to migrate to Numpy 2
- we check with @robertapplin if quasielasticbayes can be migrated during the 6.12 sprint
- we migrate matplotlib version to 3.8.X or 3.9.X
- from this point, there should not be any unsolvable difficulty left

robertapplin, 2024-09-04:
I believe quasielasticbayes (v0.2.2) should now work with Numpy 2 https://anaconda.org/conda-forge/quasielasticbayes

perenon, 2024-09-12:
> I believe quasielasticbayes (v0.2.2) should now work with Numpy 2 https://anaconda.org/conda-forge/quasielasticbayes

Thanks @robertapplin !

Tested with:
- python 3.10.14
- numpy 2.1.1
- matplotlib 3.8.4
- quasielasticbayes 0.2.2

=> Mantid workbench launches
=> 
```
FAIL: test_QSe_Run (BayesQuasiTest.BayesQuasiTest)
Test Stretched Exponential fit for BayesQuasi
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/perenon/_Dev/mantid/mantill/Framework/PythonInterface/test/python/plugins/algorithms/WorkflowAlgorithms/BayesQuasiTest.py", line 79, in test_QSe_Run
    self._validate_QSe_value(result, fit_group)
  File "/home/perenon/_Dev/mantid/mantill/Framework/PythonInterface/test/python/plugins/algorithms/WorkflowAlgorithms/BayesQuasiTest.py", line 243, in _validate_QSe_value
    self.assertAlmostEqual(result.dataY(0)[0], 81.12644, delta=1e-4)
AssertionError: np.float64(81.07385699682999) != 81.12644 within 0.0001 delta (np.float64(0.05258300317001385) difference)

```
That will need further investigations. I will keep you up to date !

sf1919, 2024-10-08:
We could do with completing #37957 first.

closed 2024-11-22 by thomashampson