mantidproject/mantid #37875: Migrate from boost::optional to std::optional

peterfpeterson, 2024-08-26:
In interest of using compiler features over library features, move from [boost::optional](https://www.boost.org/doc/libs/1_65_1/libs/optional/doc/html/index.html) to [std::optional](https://en.cppreference.com/w/cpp/utility/optional) which has been in the language since C++17. Like other things that have been matured in boost before migrated to the standard, this requires changing the import from
```c
#include <boost/optional.h>
```
to
```c
#include <optional>
```
and replacing the usage from `boost::optional<T>` to `std::optional<T>`.

peterfpeterson, 2024-08-30:
@rbauststfc To finish this would require a lot more planning/coordination with you to sort out the ISIS reflectometry GUI.

rbauststfc, 2024-09-09:
Thanks for letting me know @peterfpeterson. Feel free to get in touch when you want to discuss this further.

peterfpeterson, 2025-04-02:
Thanks @MialLewis for helping find [this diff](https://github.com/mantidproject/mantid/commit/90af4d878909afd086f1f5a71e53d3629e307db5#diff-b19d888efd0b59491e1c658693bb5446d36adc44e4c26c54741cd5e0acc96bb6L47) which has a comment about the `boost::optional<boost::optional<double>>` and similar types. Looking specifically at `parseThetaOrWhitespace()`
* the inner `boost::optional<T>` is unset if the field failed to parse
* the outer `boost::optional<boost::optional<T>` is unset if the field is empty

MialLewis, 2025-04-03:
Whilst I remember, my current ready is the following:

No input provided - external (initialized), internal (uninitialized). There is no value provided, this is a valid state.
Input provided, parse is successfull - externa, l(initialized)internal (initialized). This is a value provided, this is a valid state.
Input provided, parse is unsuccessfull - external (uninitialized), no need for the internal. A valid is provided, this is an invalid state.

The external boost is `is this a valid state`, if so, the internal boost is `was anything provided`.

I can take an action on LSS at ISIS to rework this.  I think the outer boost should be replaced with a struct that holds a state and a boost::optional, which would make this much clearer.