mantidproject/mantid #37973: Error when plotting Quasi result data with errors

robertapplin, 2024-09-13:
**Reported by:** Mona Sarter

**Description**
There is a crash when attempting to plot data from a workspace which contains some negative values in the error data. This crash should be avoided. Perhaps a warning should be shown instead that some of the errors are negative and that this should be investigated why.

Need a second opinion on this: Do we then attempt to plot the data but taking the absolute positive value of the errors?

**To reproduce**
1. Open Inelastic Bayes Fitting
2. Go to Quasi tab
3. Load following sample and resolution
4. Set EMin EMax to -0.075 and 0.075
5. Click `Run`
6. Right click the output workspace ending in `_Result`
7. Choose Plot-> SPectrum with errors
8. Choose 6-8 and click Ok
9. Crashes with this error:

```
Traceback (most recent call last):
  File "C:\Users\mlc47243\Documents\mantid-dev\mantid-1\mantid\qt\applications\workbench\workbench\plugins\workspacewidget.py", line 172, in _do_plot_spectrum
    plot_from_names(names, errors, overplot, advanced=advanced)
  File "C:\Users\mlc47243\Documents\mantid-dev\mantid-1\mantid\qt\python\mantidqt\mantidqt\plotting\functions.py", line 185, in plot_from_names
    return plot(
  File "C:\Users\mlc47243\Documents\mantid-dev\mantid-1\mantid\Framework\PythonInterface\mantid\plots\plotfunctions.py", line 89, in inner_func
    return func(workspaces, *args, **kwargs)
  File "C:\Users\mlc47243\Documents\mantid-dev\mantid-1\mantid\Framework\PythonInterface\mantid\plots\plotfunctions.py", line 224, in plot
    _do_single_plot(ax, workspaces, errors, show_title, nums, kw, plot_kwargs, log_name, log_values)
  File "C:\Users\mlc47243\Documents\mantid-dev\mantid-1\mantid\Framework\PythonInterface\mantid\plots\plotfunctions.py", line 601, in _do_single_plot
    plot_fn(ws, **plot_kwargs)
  File "C:\Users\mlc47243\Documents\mantid-dev\mantid-1\mantid\Framework\PythonInterface\mantid\plots\mantidaxes.py", line 45, in wrapper
    func_value = func(self, *args, **kwargs)
  File "C:\Users\mlc47243\Documents\mantid-dev\mantid-1\mantid\Framework\PythonInterface\mantid\plots\mantidaxes.py", line 815, in errorbar
    axesfunctions.errorbar(self, normalize_by_bin_width=is_normalized, *args, **kwargs),
  File "C:\Users\mlc47243\Documents\mantid-dev\mantid-1\mantid\Framework\PythonInterface\mantid\plots\axesfunctions.py", line 289, in errorbar
    return axes.errorbar(x, y, dy, dx, *args, **kwargs)
  File "C:\Users\mlc47243\Documents\mantid-dev\mantid-1\mantid\Framework\PythonInterface\mantid\plots\mantidaxes.py", line 45, in wrapper
    func_value = func(self, *args, **kwargs)
  File "C:\Users\mlc47243\Documents\mantid-dev\mantid-1\mantid\Framework\PythonInterface\mantid\plots\mantidaxes.py", line 823, in errorbar
    errorbar_container = Axes.errorbar(self, *args, **kwargs)
  File "C:\Users\mlc47243\AppData\Local\mambaforge\envs\mantid-developer\lib\site-packages\matplotlib\__init__.py", line 1446, in inner
    return func(ax, *map(sanitize_sequence, args), **kwargs)
  File "C:\Users\mlc47243\AppData\Local\mambaforge\envs\mantid-developer\lib\site-packages\matplotlib\axes\_axes.py", line 3636, in errorbar
    raise ValueError(
ValueError: 'yerr' must not contain negative values
```

SpencerHowells, 2024-09-13:
@robertapplin the sample data is the subtraction of 2 workspaces so the problem is there. Are there negative data values?

robertapplin, 2024-09-13:
> @robertapplin the sample data is the subtraction of 2 workspaces so the problem is there. Are there negative data values?

I see, yes there are some negative values in the input sample data.

I have also found the error propagation for the EISF errors might not be completely right, thanks to some help from @RichardWaiteSTFC .

[mantid/Framework/PythonInterface/plugins/algorithms/WorkflowAlgorithms/BayesQuasi.py](https://github.com/mantidproject/mantid/blob/c100135e10239b87a486fa275ce07c1660255a34/Framework/PythonInterface/plugins/algorithms/WorkflowAlgorithms/BayesQuasi.py#L660)

Line 660 in [c100135](https://github.com/mantidproject/mantid/commit/c100135e10239b87a486fa275ce07c1660255a34)

 EISF_error = EISF_data * np.sqrt((height_error**2 / height_data**2) + (total_error / total**2)) 
I am going to investigate this too and see if I get a different formula because of the "height" dependence in the "total"

jhaigh0, 2024-09-16:
When testing this, I don't get a crash. I get one of these uncaught exception warnings that allows you to coninue

[image]

PR #37984 (2024-09-30): Fix plot crash on negative errors
### Description of work

Fix crash in the case of plotting data with negative errors.

Instead, print a warning and plot without errors.

Fixes #37973

### To test:

- Load this data with negative errors
[BCA299_Subt_QLd_Result.zip](https://github.com/user-attachments/files/17027794/BCA299_Subt_QLd_Result.zip)

- Plot Spectra with errors 6-8
- Should see a warning printed but plot opens normally.

---

### Reviewer

Please comment on the points listed below ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)).
**Your comments wil [... cut]
closed 2024-09-30 by github-actions[bot]