mantidproject/mantid #37985: Manual Testing Indirect interfaces

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Follow the directions at: 

* https://developer.mantidproject.org/Testing/Indirect/DiffractionTests.html
* https://developer.mantidproject.org/Testing/Indirect/DataReductionTests.html

robertapplin, 2024-09-18:
On Windows there is a freeze and then crash with these instructions:

1. Open Interfaces > Indirect > Data reduction

2. Make sure Instrument is set to IRIS

3. Go to the ISIS calibration

4. Enter Input Runs 26173

5. Click Run

6. This should generate a workspace with _calib at the end, delete this workspace
7. Check the Scale by factor checkbox and change the value of the scale to 0.5.
8. Click Run
9. This should generate a workspace with _calib
10. Check the Create RES box
11. Click Run
12. This should generate a workspace with _res at the end

robertapplin, 2024-09-19:
@rbauststfc I've re-tried these tests on Windows and linuxthis morning but unfortunately I can't replicate the crashes. Are you able to replicate them still on your Windows machine?

rbauststfc, 2024-09-19:
Yes, I'm afraid it does. In case it's useful, on the crash from the ISIS Diagnostics tab, it doesn't happen if I exclude step 5 from the instructions to reproduce.

I think my Windows laptop is quite low spec, so not sure if that could have anything to do with it. You could get a few Windows users to give this a try, if you think it's worth doing a bit more digging.

rbauststfc, 2024-09-19:
I've just had a go in my installation of 6.10 and I don't get crashes in either case. The first works fine and the second flashes up an error message box and logs the following in the messages pane:

```
Error in execution of algorithm TimeSlice:
Problem setting "StartWorkspaceIndex" in CropWorkspace-v1: When converting parameter "StartWorkspaceIndex": When setting value of property "StartWorkspaceIndex": Selected value -1 is < the lower bound (0)
  at line 137 in 'C:/MantidInstall/plugins/python\algorithms/WorkflowAlgorithms/TimeSlice.py'
  caused by line 215 in 'C:/MantidInstall/plugins/python\algorithms/WorkflowAlgorithms/TimeSlice.py'
  caused by line 1081 in 'C:\MantidInstall\bin\lib\site-packages\mantid\simpleapi.py'
  caused by line 970 in 'C:\MantidInstall\bin\lib\site-packages\mantid\simpleapi.py'
  caused by line 953 in 'C:\MantidInstall\bin\lib\site-packages\mantid\simpleapi.py'
Error starting batch algorithm
Got error from algorithm "TimeSlice"
Stopping batch algorithm because of execution error
```

robertapplin, 2024-09-19:
Thank you for checking! @MohamedAlmaki and @adriazalvarez can either of you replicate these crashes with your Windows machines?

rbauststfc, 2024-09-19:
Thanks @adriazalvarez, that's interesting because I've been using the standalone installer, not a Conda install. @robertapplin, how had you installed Mantid when you tested on Windows?

robertapplin, 2024-09-19:
Thanks for testing @adriazalvarez . I was using the standalone installer too. @MohamedAlmaki do you have any luck replicating this?

MohamedAlmaki, 2024-09-19:
@robertapplin I could reproduce both crashes on my Windows machine, but on Linux works fine. The crash happens in the reading file step. I'm going to investigate more with a development build

robertapplin, 2024-09-19:
> @robertapplin I could reproduce both crashes on my Windows machine, but on Linux works fine. The crash happens in the reading file step. I'm going to investigate more with a development build

Thats great, thank you!

closed 2024-09-26 by SilkeSchomann