mantidproject/mantid #37986: Manual Testing Inelastic interfaces

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Follow the directions at:

* https://developer.mantidproject.org/Testing/Inelastic/BayesFittingTests.html
* https://developer.mantidproject.org/Testing/Inelastic/CorrectionsTests.html
* https://developer.mantidproject.org/Testing/Inelastic/QENSFittingTests.html 
* https://developer.mantidproject.org/Testing/Inelastic/DataProcessorTests.html

cailafinn, 2024-09-19:
Possible typo in  Bayes -> ResNorm testing instructions. 
`irs26173_graphite002_red` is missing from the Sample Data. 
`irs26176_graphite002_red` is present and is the one used in all other tests.

cailafinn, 2024-09-19:
Deprecation warning when running on Quasi tab: This algorithm is deprecated, please use BayesQuasi2 instead.

warunawickramasingha, 2024-09-19:
[Quasi Tab](https://developer.mantidproject.org/Testing/Inelastic/BayesFittingTests.html#id3) step 12: the titles of the plots does not reflect what is being plotted as to whether Prob, Amplitude and FWHM

warunawickramasingha, 2024-09-19:
https://developer.mantidproject.org/Testing/Inelastic/BayesFittingTests.html#id4 step 12. It only opens two plots, not `three` plots

warunawickramasingha, 2024-09-19:
https://developer.mantidproject.org/Testing/Inelastic/CorrectionsTests.html#id4 step:7 "This should plot a blue **corrected** line"

warunawickramasingha, 2024-09-19:
https://developer.mantidproject.org/Testing/Inelastic/DataProcessorTests.html#id3 step 18: instruction need to be updated to "On **Spectrum Numbers** select 5"

RichardWaiteSTFC, 2024-09-19:
On [Moments tab](https://developer.mantidproject.org/Testing/Inelastic/DataProcessorTests.html#id5)
After final step of Plot Spectra, the line edit is disabled and subsequent plots with different spectrum indices cannot be made unless you click Run again

warunawickramasingha, 2024-09-19:
> On [Moments tab](https://developer.mantidproject.org/Testing/Inelastic/DataProcessorTests.html#id5) After final step of Plot Spectra, the line edit is disabled and subsequent plots with different spectrum indices cannot be made unless you click Run again

This is Not a regression.

warunawickramasingha, 2024-09-19:
> Crash in [Symmetrise Tab](https://developer.mantidproject.org/Testing/Inelastic/DataProcessorTests.html#symmetrise-tab) when:
> 
> 1. Data is loaded
> 2. ADS is cleared
> 3. Preview button is pressed
> 
> THIS IS A REGRESSION. Not present in 6.10, should be fixed for 6.11. Seen on macOS and IDAaaS.

https://github.com/mantidproject/mantid/issues/38042 was raised!

yusufjimoh, 2024-09-19:
Incorrect plot label for on Quasi and Stretch Tab 
#38043 has been created for it

warunawickramasingha, 2024-09-20:
https://github.com/mantidproject/mantid/issues/38064 was raised to update test instructions as mentioned above

warunawickramasingha, 2024-09-20:
> On [Moments tab](https://developer.mantidproject.org/Testing/Inelastic/DataProcessorTests.html#id5) After final step of Plot Spectra, the line edit is disabled and subsequent plots with different spectrum indices cannot be made unless you click Run again

https://github.com/mantidproject/mantid/issues/38065 was raised

warunawickramasingha, 2024-09-20:
Test time ~1.50mints

> Deprecation warning when running on Quasi tab:
> 
> ```
> This algorithm is deprecated, please use BayesQuasi2 instead.
> ```

https://github.com/mantidproject/mantid/issues/38066 raised.

warunawickramasingha, 2024-09-20:
> Deprecation warning when running on Quasi tab:
> 
> ```
> This algorithm is deprecated, please use BayesQuasi2 instead.
> ```

https://github.com/mantidproject/mantid/issues/38066 was raised

warunawickramasingha, 2024-09-20:
Test time~1h and 50mints

closed 2024-09-20 by warunawickramasingha