mantidproject/mantid #37988: Manual Testing Workbench Core

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/Core/Core.html

rbauststfc, 2024-09-18:
When clicking on question mark on a Superplot it links to the docs on Basic 1D and Tiled plots but might be good if it linked to the Superplot docs: https://docs.mantidproject.org/nightly/workbench/superplot.html?

robertapplin, 2024-09-18:
Loading this data and then doing a colorfill plot, I get an error https://stfc365-my.sharepoint.com/:t:/r/personal/robert_applin_stfc_ac_uk/Documents/Documents/WV_96538.txt?csf=1&web=1&e=TAOVRO
```
Traceback (most recent call last):
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/workbench/plugins/workspacewidget.py", line 217, in _do_plot_colorfill
    pcolormesh(names)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/mantid/plots/plotfunctions.py", line 89, in inner_func
    return func(workspaces, *args, **kwargs)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/mantidqt/plotting/functions.py", line 294, in pcolormesh
    fig.canvas.draw()
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/matplotlib/backends/backend_agg.py", line 400, in draw
    self.figure.draw(self.renderer)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/matplotlib/artist.py", line 95, in draw_wrapper
    result = draw(artist, renderer, *args, **kwargs)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/matplotlib/artist.py", line 72, in draw_wrapper
    return draw(artist, renderer)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/matplotlib/figure.py", line 3175, in draw
    mimage._draw_list_compositing_images(
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/matplotlib/image.py", line 131, in _draw_list_compositing_images
    a.draw(renderer)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/matplotlib/artist.py", line 72, in draw_wrapper
    return draw(artist, renderer)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/matplotlib/axes/_base.py", line 3064, in draw
    mimage._draw_list_compositing_images(
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/matplotlib/image.py", line 131, in _draw_list_compositing_images
    a.draw(renderer)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/matplotlib/artist.py", line 39, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/mantid/plots/resampling_image/samplingimage.py", line 75, in draw
    self._resample_image()
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/mantid/plots/resampling_image/samplingimage.py", line 111, in _resample_image
    x, y, data = get_matrix_2d_ragged(
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/mantid/plots/datafunctions.py", line 542, in get_matrix_2d_ragged
    x_centers = mantid.plots.datafunctions.points_from_boundaries(x_edges)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/mantid/plots/datafunctions.py", line 183, in points_from_boundaries
    raise ValueError("could not get centers from less than two boundaries")
ValueError: could not get centers from less than two boundaries
```

robertapplin, 2024-09-18:
Script to file for tiled plot produces a file with multiple of the same imports

robertapplin, 2024-09-18:
Create a md_4D workspace
```py
md_4D = CreateMDWorkspace(Dimensions=4, Extents=[-1,1,-1,1,-1,1,-10,10], Names="H,K,L,E", Units="U,U,U,V")
FakeMDEventData(InputWorkspace=md_4D, PeakParams='500000,0,0,0,0,3')
```
Do Show sample logs for md_4D workspace and you get a warning:
```
Could not open sample logs for workspace 'md_4D'.
ValueError: MDWorkspace::getExperimentInfo(): expInfoIndex is out of range.
```

robertapplin, 2024-09-18:
Showing Instrument for the WV_96538 dataset gives an error:
https://stfc365-my.sharepoint.com/:t:/r/personal/robert_applin_stfc_ac_uk/Documents/Documents/WV_96538.txt?csf=1&web=1&e=TAOVRO
```
In Instrument::getSamplePos(). No SamplePos has been set.
Error in execution of algorithm CreateDetectorTable:
Matrix workspace has no instrument information
```

robertapplin, 2024-09-18:
Load MAR11060
Plot Spectrum 4
Select Fit
Change Minimizer to "Simplex"
Select Fit->Evaluate Function
There is a segmentation fault

SilkeSchomann, 2024-09-20:
None of the issues described above is a regression when compared to Mantid 6.10. I will raise separate GitHub issues for each, but do not think any of them need to be fixed urgently for the upcoming release.

closed 2024-09-26 by SilkeSchomann