mantidproject/mantid #37989: Manual Testing Muon

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

See instructions at: https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html

MohamedAlmaki, 2024-09-18:
small typo in "Inspect the flat background by using the arrorws next to Display Parameters For, all but one should be non-zero", should be arrows

RichardWaiteSTFC, 2024-09-18:
EMU [Ionic Diffusion Test](https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/Muon_EMU.html#id2)
External plot should exclude legend from tight_layout calculation as plot size really small

RichardWaiteSTFC, 2024-09-18:
Still got ALC and FDA tests to do tomorrow 10.30

RichardWaiteSTFC, 2024-09-19:
FDA Test MaxEnt - clicking cancel shows a poop-up with keyboard interrupt and lots of error text which is a bit off-putting! Could be improved!

RichardWaiteSTFC, 2024-09-19:
Testing time: 2h30m

RichardWaiteSTFC, 2024-09-19:
Closing as all issues raised - all for v6.12 IMO

closed 2024-09-19 by RichardWaiteSTFC