mantidproject/mantid #37990: Manual Testing TOFConverter

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

Make sure that inputs and outputs work sensibly, stress test with some bad inputs (e.g. letters in a numeric input)
https://docs.mantidproject.org/interfaces/utility/TOF%20Converter.html

MohamedAlmaki, 2024-09-18:
In a TOF converter, there is no limit to the scattering angle

adriazalvarez, 2024-09-18:
Clicking help button on Conda installation crashes Mantid

RichardWaiteSTFC, 2024-09-18:
Interface does not have ADS observers so:
- Does not recognise any single valued workspaces created after opening (not a big deal just close/open again)
- Seems to read the value in workspaces on opening, not on convert, so if you replace the workspace or alter the value in-place using `setY` it uses the wrong number - again perhaps not unexpected. But could read the number on Convert being clicked and update the box on the itnerface.

adriazalvarez, 2024-09-18:
Finish time: 15 mins

closed 2024-09-18 by adriazalvarez

sf1919, 2024-09-18:
Do any of these need writing up as issues? I know the Windows help one is #34739

RichardWaiteSTFC, 2024-09-18:
I don't think they can be regressions as this interface hasn't been touched yet has right? I think adding ADS observers is maybe a bit overboard in terms of making it more complex for little gain. The behaviour though perhaps sub-optimal is at least self-consistent (i.e. snapshot of workspaces as they were when you open it)