mantidproject/mantid #37992: Manual Testing Engineering Diffraction

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/EngineeringDiffraction/EngineeringDiffractionTestGuide.html

Despiix, 2024-09-18:
Engineering Diffraction Analysis:
During test 1 (Calibration) when hovering over the plots I got this error.

Traceback (most recent call last):
  File "C:\MantidNightlyInstall\bin\lib\site-packages\matplotlib\cbook\__init__.py", line 314, in process
    func(*args, **kwargs)
  File "C:\MantidNightlyInstall\bin\lib\site-packages\workbench\plotting\figureinteraction.py", line 141, in on_scroll
    self._correct_for_scroll_event_on_legend(event)
  File "C:\MantidNightlyInstall\bin\lib\site-packages\workbench\plotting\figureinteraction.py", line 174, in _correct_for_scroll_event_on_legend
    if legend.get_draggable() and legend.contains(event):
AttributeError: 'NoneType' object has no attribute 'get_draggable'

jhaigh0, 2024-09-18:
Focus step of test 1. Can't find the vanadium run 307521. (IDAaaS)

jclarkeSTFC, 2024-09-18:
I'm having problems using the archive on macOS so I can't follow the testing instructions

jhaigh0, 2024-09-18:
> Engineering Diffraction Analysis: During test 1 (Calibration) when hovering over the plots I got this error.
>
> Traceback (most recent call last): File [...]

I can recreate this when scrolling on the plot

Despiix, 2024-09-18:
Cannot follow test instructios, red `*` will not dissapear.

jhaigh0, 2024-09-18:
Loading the data on idaaas via the run numbers is too unreliable, can't perform the manual tests

jhaigh0, 2024-09-18:
Issue for scrolling created
https://github.com/mantidproject/mantid/pull/38015

GuiMacielPereira, 2024-09-18:
Noticed that on IDAaaS, the data cache search is still failing, so re-opened https://github.com/mantidproject/mantid/issues/37398

GuiMacielPereira, 2024-09-19:
I followed though the rest of the instructions and had no further issues. Took some time to install GSASII on Linux due to timeout errors. Didn't keep track of time as I did it over several smaller sessions but I estimate it to be between 1h15 - 1h30.

Despiix, 2024-09-19:
Finished testing except, the last test because I couldn't find the correct GSAS-II path. Time ~2hrs

closed 2024-09-20 by GuiMacielPereira