mantidproject/mantid #37994: Manual Testing ISIS Reflectometry

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

RichardWaiteSTFC, 2024-09-18:
1. Load Batch File Test > Not clear what/where Expand group button is - perhaps a picture would help?
2. Add/Delete Batch Test > Deleting all batch tabs, new one opens but keeps iterating number

MohamedAlmaki, 2024-09-18:
I got a crash when trying to change the polarization correction option in the experiment settings from FilePath to ParameterFile. We couldn't reproduce again so it might be just a random crash. Just a note that there were multiple load plugin failures in the stack trace but nothing related to the reflectometry interface

adriazalvarez, 2024-09-18:
In search by experiment instructions: Whenever you try to transfer runs to the runs table, if there is an empty group on the table, the transferred run won't override this empty group but be placed in a row under the empty group.  Additionally, if you have added 2 or 3 groups to the table, then you add an empty group, and then you transfer more runs, the empty group will stay there in the table. Empty groups should disappear when transferring data.

adriazalvarez, 2024-09-18:
When changing data on a run row (for example the angle). it adds a new empty row just below.

adriazalvarez, 2024-09-18:
There are many (long) instructions, particularly in the last sections of the manual test. I would be helpful for following the instructions to have them numbered.

adriazalvarez, 2024-09-18:
On the preview tab instructions, the following instruction doesn't match the results on the test, there is no ReflectometryISISSumBanks property on the history of the workspace: 

```
Right-click the workspace and select Show History:
    In the Algorithms list, expand ReflectometryISISLoadAndProcess.
    Click on ReflectometryReductionOneAuto and check in the right hand pane that the inputs for ProcessingInstructions, BackgroundProcessingInstructions and TransmissionProcessingInstructions correspond to the ranges of spectra you selected.
    Expand ReflectometryReductionOneAuto.
    Click on ReflectometryISISSumBanks and check that the input for ROIDetectorIDs matches the range of detector IDs you selected.

```