mantidproject/mantid #37995: Manual Testing ISIS SANS

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/SANSGUI/ISISSANSGUITests.html

rbauststfc, 2024-09-18:
Clarify section 1 instructions and where to put settings back to at end.

rbauststfc, 2024-09-18:
Add instructions to clear workspaces list at start of Processing section.

rbauststfc, 2024-09-18:
2D reduction, need to change reduction mode back to All and need to remove instructions about plot results checkbox.

rbauststfc, 2024-09-18:
Background subtraction instructions need some updates - saving to file and no need to edit user file reduction mode any more?

robertapplin, 2024-09-18:
Beam centre finder test gave this error:
```
Error in execution of algorithm SANSBeamCentreFinder:

  at line 147 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
  caused by line 192 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
  caused by line 240 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
There has been an error. See more: RuntimeError on line 82 of '/home/mlc47243/mambaforge/envs/mantid_env/scripts/SANS/sans/algorithm_detail/centre_finder_new.py': 
  at line 147 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
  caused by line 192 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
  caused by line 240 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
```

rbauststfc, 2024-09-19:
> Beam centre finder test gave this error:
>
> ```
> Error in execution of algorithm SANSBeamCentreFinder:
>
>   at line 147 in [...]

@robertapplin I've had a go at reproducing this today, both on Windows and IDAaaS, and I can't seem to. I'm just going straight into the beam centre finder test rather than completing all of the steps leading up to it, though. Are you able to reproduce this today at all?

Looking at the line in the code where the error originated, it's an error that's thrown if the workspace that is trying to be plotted contains a `NaN` value. This isn't a regression, it looks like there was some refactoring several years ago that changed the code to throw an error in these circumstances. Before that change it would have just stopped the plotting but continue with the rest of the process. It's not clear why that change was made. I will open an issue to investigate this further at some point, but not something that needs looking at for 6.11.

rbauststfc, 2024-09-19:
I've created an issue for updating the SANS GUI manual testing instructions - #38025. I'll add more details to that issue and put it on the Mantid Maintenance board in due course.

rbauststfc, 2024-09-19:
I'm going to close this because we've finished the testing. I've opened issues for the relevant comments, so any further information can now be added to those issues if necessary.

closed 2024-09-19 by rbauststfc

robertapplin, 2024-09-19:
> @robertapplin I've had a go at reproducing this today, both on Windows and IDAaaS, and I can't seem to. I'm just going straight into the beam centre finder test rather than completing all of the steps leading up to it, though. Are you able to reproduce this today at all?

I've just tried replicating on Windows, Linux and IDAaaS with no luck. I also tried running the tests before the beam centre finder test, and that still didn't cause the same problem I saw yesterday. Something I did spot is when I change the run in the first column of the runs table to `74045`, and then do the beam centre finder test, I get this error:

```
Error in execution of algorithm CropToComponent:
SpectrumDefinition contains multiple entries. No unique mapping from detector to spectrum possible
Error in execution of algorithm SANSBeamCentreFinderCore:
SpectrumDefinition contains multiple entries. No unique mapping from detector to spectrum possible
  at line 149 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinderCore.py'
  caused by line 300 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinderCore.py'
Error in execution of algorithm SANSBeamCentreFinder:
SpectrumDefinition contains multiple entries. No unique mapping from detector to spectrum possible
  at line 149 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinderCore.py'
  caused by line 300 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinderCore.py'
  at line 147 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
  caused by line 177 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
  caused by line 243 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
  caused by line 364 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
There has been an error. See more: RuntimeError on line 82 of '/home/mlc47243/mambaforge/envs/mantid_env/scripts/SANS/sans/algorithm_detail/centre_finder_new.py': SpectrumDefinition contains multiple entries. No unique mapping from detector to spectrum possible
  at line 149 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinderCore.py'
  caused by line 300 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinderCore.py'
  at line 147 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
  caused by line 177 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
  caused by line 243 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
  caused by line 364 in '/home/mlc47243/mambaforge/envs/mantid_env/plugins/python/algorithms/WorkflowAlgorithms/SANS/SANSBeamCentreFinder.py'
```
but this might be expected?

rbauststfc, 2024-09-19:
Unfortunately I'm not up to speed enough yet with the beam centre finder to know exactly what scenarios it's designed to handle and what would be unrealistic uses - we've not done any work in this area since I joined LSS. However, I think it would probably be expected for it to give errors depending on whether you were passing it suitable data and/or settings. It seems to me that the errors could maybe be more helpful/user friendly, but I don't think this is a regression and not something users have complained about. So I'll link to your comment in the issue I've opened, but I don't think we need to worry about that one for 6.11. Hopefully once we've looked at issue #38026 we'll have a better understanding of the feature and can address any usability issues with error messages, etc.