mantidproject/mantid #37996: Manual Testing Live Data ISIS

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/LiveData/LiveDataTests.html

jhaigh0, 2024-09-18:
Using a `dev.analysis.stfc.ac.uk` mantid developer idaaas workspace, my `Mantid.user.properties` file is read only, so I could not complete tests which required the ADARA Fake Event setup. Everythinh else worked fine.

GuiMacielPereira, 2024-09-18:
Everything worked fine, took around 20 minutes.

closed 2024-09-20 by GuiMacielPereira