mantidproject/mantid #37997: Manual Testing Mantid Basics Course

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

* Check the Qt-help docs, built into Workbench, against the online documentation 
* Open up the basic course: http://docs.mantidproject.org/tutorials/mantid_basic_course/index.html#mantid-basic-course
* Check that the information and exercises are up to date and work in the Workbench release candidate

cailafinn, 2024-09-18:
Plot options menu needs an updated screenshot.

cailafinn, 2024-09-18:
"Remove Peaks" button will remove the nearest peak, even if it is not currently on the screen but is "closer" in space after L value (for example) is changed.

cailafinn, 2024-09-18:
Testing instructions: 
Screenshot of cutviewer plot (point 8) seems to now be different. This should probably be checked that it's still the right output (or at least have the screenshot in the documentation updated).

cailafinn, 2024-09-18:
Is regression?

X axis label has 0.1 in H position.

warunawickramasingha, 2024-09-20:
> Plot options menu needs an updated screenshot. Others are from Windows so should probably stay matching.

https://github.com/mantidproject/mantid/issues/38050 was raised to update the screeshot

warunawickramasingha, 2024-09-20:
Testing time ~1h

closed 2024-09-20 by warunawickramasingha