mantidproject/mantid #37999: Manual Testing Project Recovery

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

robertapplin, 2024-09-18:
Test 3 does not work

```
ValueError: Problem setting "RHSWorkspace" in MultiplyMD-v1: Invalid value for property RHSWorkspace (IMDWorkspace) from string "__python_binary_op_single_value": Workspace "__python_binary_op_single_value" was not found in the Analysis Data Service
  File "/home/mlc47243/.mantid/ordered_recovery.py", line 20, in <module>
    MultiplyMD(LHSWorkspace='long4', RHSWorkspace='__python_binary_op_single_value', OutputWorkspace='long4') # 2024-09-18T12:45:48.838808000 execCount: 44
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/mantid/simpleapi.py", line 1082, in __call__
    set_properties(algm, *args, **final_keywords)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/mantid/simpleapi.py", line 971, in set_properties
    do_set_property(key, value)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/mantid/simpleapi.py", line 954, in do_set_property
    raise e.__class__(msg) from e
```

robertapplin, 2024-09-18:
In testing instruction 6, the `Only open in script editor` text should be `Just open in script editor`

closed 2024-09-18 by robertapplin