mantidproject/mantid #38001: Manual Testing Sliceviewer

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html

perenon, 2024-09-18:
On Linux, conda installation.

Could not perform completely the test "Basic Usages => MDWorkspace", since setUB algorithm could not be imported:

At startup:
```
ImportError: cannot import name 'SetUB' from partially initialized module 'mantid.simpleapi' (most likely due to a circular import) (/home/perenon/miniconda3/envs/mantid_env/lib/python3.10/site-packages/mantid/simpleapi.py)`
```

At runtime:
```
NameError: name 'SetUB' is not defined
  File "/home/perenon/mari.py", line 10, in <module>
```

Seems related to https://github.com/mantidproject/mantid/issues/38011

warunawickramasingha, 2024-09-18:
https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#id8 the screenshot is outdated

warunawickramasingha, 2024-09-18:
https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#id8 the plot at point 8 is different to what is shown in the docs
[image]

warunawickramasingha, 2024-09-18:
https://docs.mantidproject.org/nightly/interfaces/general/Fit%20Script%20Generator.html#id8 step 12: the changes seems to be applied on all the other domain functions by index position. Need to check if this is a regression or the same behaviour.

cailafinn, 2024-09-19:
Occasional error when closing the sliceviewer:
```log
Traceback (most recent call last):
  File "/Users/caila.finn/mambaforge/envs/mantid_6.11/lib/python3.10/site-packages/matplotlib/cbook/__init__.py", line 314, in process
    func(*args, **kwargs)
  File "/Users/caila.finn/mambaforge/envs/mantid_6.11/lib/python3.10/site-packages/mantidqt/widgets/sliceviewer/views/cursor.py", line 57, in on_mouse_leave
    self.on_cursor_outside_axes()
  File "/Users/caila.finn/mambaforge/envs/mantid_6.11/lib/python3.10/site-packages/mantidqt/widgets/sliceviewer/presenters/imageinfowidget.py", line 67, in on_cursor_outside_axes
    self._widget.cursorAt(DBLMAX, DBLMAX, DBLMAX, extra_cols)
RuntimeError: wrapped C/C++ object of type ImageInfoWidget has been deleted
```

warunawickramasingha, 2024-09-20:
[image]
 https://github.com/mantidproject/mantid/issues/38056 was raised for this.

warunawickramasingha, 2024-09-20:
https://github.com/mantidproject/mantid/issues/37997#issuecomment-2358803302

This was working fine in IDAaaS both in 6.10 and in the version used for 6.11 manual testing. @cailafinn could you please double check and raise an issue if necessary? 
Thanks

RichardWaiteSTFC, 2024-10-03:
Finished testing basic usage for MD workspaces, Peak overlay, Cutviewer and "specific Tests" on IDAaaS  using nightly from 30/09 conda installed.

Updates to docs (will post in #38056):
- Cutviewer test step 19 - need to move marker to L~2 to see peak at x=-1
- Cutviewer step 24 - clarify that H integration width should be doubled in tthe spin box above the colorfill plot (row of integrated dimension of slice is not editable table above cut viewer)

Only other comment is that cutviewer calls BinMD a lot which fills up the log window - perhaps it should be called with EnableLogging=False.

warunawickramasingha, 2024-10-03:
Closing this as testing is complete and https://github.com/mantidproject/mantid/issues/38056 was raised to update the testing instructions.

closed 2024-10-03 by warunawickramasingha