mantidproject/mantid #38002: Manual Testing Fit Script Generator

MohamedAlmaki, 2024-09-17:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Disable Usage reporting before you start testing
* Close the this issue once you are done.
* Time how long this manual test takes for you to do and leave a comment about it in this issue.

### Specific Notes:

https://docs.mantidproject.org/nightly/interfaces/general/Fit%20Script%20Generator.html

warunawickramasingha, 2024-09-18:
update instructions to where to find Fit Script Generator interface

cailafinn, 2024-09-18:
Making changes to function parameters doesn't work as expected if one of the domains has more functions in it than the others. Visible in instruction 12 [here](https://docs.mantidproject.org/nightly/interfaces/general/Fit%20Script%20Generator.html#usage-example-for-simultaneous-fitting)

warunawickramasingha, 2024-09-18:
https://docs.mantidproject.org/nightly/interfaces/general/Fit%20Script%20Generator.html#id8 step 12: the changes seems to be applied on all the other domain functions by index position. Need to check if this is a regression or the same behaviour.

cailafinn, 2024-09-18:
> Making changes to function parameters doesn't work as expected if one of the domains has more functions in it than the others. Visible in instruction 12 [here](https://docs.mantidproject.org/nightly/interfaces/general/Fit%20Script%20Generator.html#usage-example-for-simultaneous-fitting)

Not a regression.

cailafinn, 2024-09-18:
Issue created at
- #38021

warunawickramasingha, 2024-09-20:
Closing this since tickets were created for all identified issues.

closed 2024-09-20 by warunawickramasingha