mantidproject/mantid #38011: Module load error on IDAaaS

jclarkeSTFC, 2024-09-17:
**Describe the bug**
Certain plugins on IDAaaS generate an error on starting Workbench, and some scripts won't run, the error below is produced

```
Welcome to Mantid 6.10.20240916.1815
Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
Could not open library /opt/mantidworkbenchnightly/plugins/libMantidCrystal.so: libMantidDataHandling.so: cannot open shared object file: No such file or directory
Failed to load plugin /opt/mantidworkbenchnightly/plugins/python/algorithms/BASISCrystalDiffraction.py.
Error: Traceback (most recent call last):
  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantid/kernel/plugins.py", line 198, in load_from_file
    name, module = load_plugin(filepath)
  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantid/kernel/plugins.py", line 219, in load_plugin
    module = loader.run()
  File "/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantid/kernel/plugins.py", line 52, in run
    loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/opt/mantidworkbenchnightly/plugins/python/algorithms/BASISCrystalDiffraction.py", line 28, in <module>
    from mantid.simpleapi import (
ImportError: cannot import name 'SetGoniometer' from partially initialized module 'mantid.simpleapi' (most likely due to a circular import) (/opt/mantidworkbenchnightly/lib/python3.10/site-packages/mantid/simpleapi.py)
```

**To Reproduce**
Start the nightly on IDAaaS

**Expected behavior**
No errors

**Screenshots**

[image]

**Platform/Version (please complete the following information):**
 - OS: Rocky Linux
 - OS Version:
 - Mantid Version 6.10.20240916.1815

rbauststfc, 2024-09-18:
Just to flag, this happened for me on a Windows install.

perenon, 2024-09-18:
Happened also to me on Linux conda install

jclarkeSTFC, 2024-09-25:
This is not fixed

PR #38110 (2024-10-01): Revert #37908
That PR (#37908) causes a dynamic library loading problem in the release. I tried to fix it by changing the new dependency from `PRIVATE` to `PUBLIC` but that didn't work.

Fixes #38011. 

### To test:

The problem only appears when using a standalone installer, not a conda install (don't know why). [This](https://builds.mantidproject.org/job/Core_Team_test_pipeline/118/artifact/) will give you a Windows installer until the full `build_packages_from_branch` can run for the other operating systems.

https://builds.mantidproject.org/job/build_packages_from_branch/908/

---

### Rev [... cut]
closed 2024-10-01 by github-actions[bot]