mantidproject/mantid #38018: Project Recovery Test 3 fails with an error

robertapplin, 2024-09-18:
**Describe the bug**
There is an error when following the Project Recovery testing instructions for test 3. I think this is related to a change made with regards to `StoreInADS` recently.

**To Reproduce**
Test 3 in the following instructions fails with an error. https://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ProjectRecoveryTesting.html

```
ValueError: Problem setting "RHSWorkspace" in MultiplyMD-v1: Invalid value for property RHSWorkspace (IMDWorkspace) from string "__python_binary_op_single_value": Workspace "__python_binary_op_single_value" was not found in the Analysis Data Service
  File "/home/mlc47243/.mantid/ordered_recovery.py", line 20, in <module>
    MultiplyMD(LHSWorkspace='long4', RHSWorkspace='__python_binary_op_single_value', OutputWorkspace='long4') # 2024-09-18T12:45:48.838808000 execCount: 44
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/mantid/simpleapi.py", line 1082, in __call__
    set_properties(algm, *args, **final_keywords)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/mantid/simpleapi.py", line 971, in set_properties
    do_set_property(key, value)
  File "/home/mlc47243/mambaforge/envs/mantid_env/lib/python3.10/site-packages/mantid/simpleapi.py", line 954, in do_set_property
    raise e.__class__(msg) from e
```

**Expected behavior**
The recovery script should not fail with an error message.

**Platform/Version (please complete the following information):**
All operating systems

robertapplin, 2024-09-18:
@jhaigh0 might know best why this is happening

jhaigh0, 2024-09-18:
yeah I assume I've caused this here https://github.com/mantidproject/mantid/pull/37839

PR #38037 (2024-09-19): Remove workspace history workaround from binaryOpertaions
### Description of work

Previously, there was a workaround in `BinaryOperations::performBinaryOpWithDouble` to ensure that workspace history was preserved. This involved adding a workspace briefly to the ads in order to give it a `name` property.

Since the pr https://github.com/mantidproject/mantid/pull/37839, algorithms run with `storeInADS=False` should have their histories reconstructed correctly, and not need such workarounds.

I think what's now happening in the function is covered by tests added in the previous pr.

Fixes #38018 [... cut]
closed 2024-09-19 by github-actions[bot]