mantidproject/mantid #38056: Update test instructions in SliceViewer

warunawickramasingha, 2024-09-20:
During manual testing for Mantid 6.11 it was seen that the screenshot used in [CutViewer Tool](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#id8) at `step 8` seems outdated. 

It needs to double check whether it was just an outdated screenshot or the observed output as shown below was correct.

[image]

**Describe any solutions you are considering**

**Additional context**

RichardWaiteSTFC, 2024-10-03:
I can confirm the screenshot from step 8 needs updating - there was a bug in the FakeMDEventData algorithm that was fixed in #33602

Some additional changes required:
- Cutviewer test step 19 - need to move marker to L~2 to see peak at x=-1
- Cutviewer step 24 - clarify that H integration width should be doubled in tthe spin box above the colorfill plot (row of integrated dimension of slice is not editable table above cut viewer)

warunawickramasingha, 2025-01-23:
> * Cutviewer test step 19 - need to move marker to L~2 to see peak at x=-1

At step 18 the marker has already dragged L ~ 2. Do you think it is worth re-mentioning that in step 19?

RichardWaiteSTFC, 2025-01-24:
> > * Cutviewer test step 19 - need to move marker to L~2 to see peak at x=-1
> 
> At step 18 the marker has already dragged L ~ 2. Do you think it is worth re-mentioning that in step 19?

From the instructions here
https://github.com/mantidproject/mantid/blob/77501cbee17678a70413a00853a977097f9d60fc/dev-docs/source/Testing/SliceViewer/SliceViewer.rst?plain=1#L321-L326

I would say it's not required to mention `L~2` in step 19 do you agree?

RichardWaiteSTFC, 2025-01-24:
> > * Cutviewer step 24 - clarify that H integration width should be doubled in tthe spin box above the colorfill plot (row of integrated dimension of slice is not editable table above cut viewer)
> 
> [@RichardWaiteSTFC](https://github.com/RichardWaiteSTFC) Is below the value you mean here? I can only see this transition when slice thickness along H is changed from 0.1 -> 0.2. It was not exactly doubled though.

The spin box to define the H integratio nwidth is at the top to the right of the cursor info table.
The counts doubling refers to the height of the peak in the 1D.
[image]

warunawickramasingha, 2025-01-24:
Thanks @RichardWaiteSTFC for the clarifications! I have already included the update to `Cutviewer step 24` as you have suggested.

closed 2025-02-13 by cailafinn