mantidproject/mantid #38062: Mantid Help crashes on Linux

GuiMacielPereira, 2024-09-20:
**Describe the bug**
Mantid crashes everytime on Linux when the help page is clicked.
Marked as a regression since this was not present in Mantid 6.9

**To Reproduce**
Steps to reproduce the behavior:
Installed standalone mantid by downloading and unziping the latest mantid release. 
Upon opening, go to Help -> Mantid Help, mantid crashes everytime with:

```
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: iris
MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: iris
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
Fontconfig error: Cannot load default config file: No such file: (null)
```

**Expected behavior**
Even if this is not completely fixed, there should at least be no crash.

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: Pop_OS 22.04 (very similar to Ubuntu 22.04)
 - OS Version: 22.04
 - Mantid Version 6.11

**Additional context**

It's known that the first time that help is clicked it tends to display a missing page error, so this issue can be considered fixed even if that is the behaviour.

GuiMacielPereira, 2024-09-25:
Opened https://github.com/mantidproject/mslice/issues/1001 on mslice repo

thomashampson, 2024-10-16:
For me the help opens fine on Ubuntu 20.04
It also works on IDAaaS

jhaigh0, 2024-10-17:
On all instances where we have the standalone launching (IDAaaS and Tom's Ubuntu) the help is opening normally.

closed 2024-10-17 by jhaigh0