mantidproject/mantid #38079: Unexpected exception when run ISIS SANS beam centre finder with no direction selected

rbauststfc, 2024-09-25:
**Original reporter:** Dirk from the SANS Group at ISIS

**Describe the bug**

If you run the beam centre finder in the ISIS SANS GUI without any direction selected then message "Have chosen no find direction exiting early" is printed in the dialog window and Mantid throws an unexpected exception (pops up the terminate dialog box). The stack trace is:

```
  File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/async_workers/beam_centre_async.py", line 47, in success_cb_slot
    self._parent_presenter.on_update_centre_values(result.output)
  File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/presenter/beam_centre_presenter.py", line 60, in on_update_centre_values
    self._beam_centre_model.update_centre_positions(new_vals)
  File "/opt/mantidworkbenchnightly/scripts/SANS/sans/gui_logic/models/beam_centre_model.py", line 58, in update_centre_positions
    self._rear_pos_1 = results["pos1"]
TypeError: 'NoneType' object is not subscriptable
```

**To Reproduce**

1) Open the ISIS SANS GUI
1) Use the `Load User File` button to load `MaskFile.toml` from `TrainingCourseData->loqdemo`
1) Use the `Load Batch File` button to load `batch_mode_reduction.csv` from `TrainingCourseData->loqdemo`
1) Click the `Beam Centre` option from the left hand menu to open the beam centre tab.
1) Under the Direction settings, untick the `Left/Right` checkbox **and** the `Up/Down` checkbox, so that nothing has been selected in this section.
1) Click the `Run` button.
1) Error "Have chosen no find direction exiting early" will be printed in red in the message pane within the tab of the ISIS SANS GUI. An unexpected exception will also be thrown.

**Expected behavior**
Needs a discussion with scientists to confirm what the appropriate behaviour should be here. If it's correct that selecting a direction should be enforced then we should probably change this to be a radio button control.

**Platform/Version (please complete the following information):**
 - OS: reproduced on Linux and Windows
 - Mantid Version [e.g. 6.0.0]

**Additional context**
This came through as two error reports, with the following descriptions:

- Trying to find the beam center. First run was with 0 iterations, just to get an idea. Setting to 20 iterations to start the search it complains that "Have chosen no find direction exiting early". At least Mantid seems to have taken a conscious decision ;->, I accept that, but we should have a chat about the reason.
- It would help if the error would say "have you ticked a direction?", or so.

smk78, 2024-09-26:
I do not understand this issue!
I have just opened 6.10.0  to process LOQ data and there are no circumstances under which I can untick BOTH Find Front Centre and Find Rear Centre! One or other of the radio buttons must be checked.
How did you untick both?

rbauststfc, 2024-09-26:
@smk78, it's these checkboxes (circled in red) that you want to be looking at:

[image]

smk78, 2024-09-26:
Oh, sorry! I should read issues more carefully!
But what we're saying is someone made a conscious decision to untick both boxes! I think we just make it that at least one box always needs to be checked, right?

smk78, 2024-09-26:
@dehoni comments by email:
>The text of the error message is a bit cryptic. And is it really an error? It should be a warning or a notice like: "no direction selected for beam centre finder." 
This hint should already be sufficient to guide a user.

rbauststfc, 2024-09-26:
@dehoni, I think the error log level was probably selected in this case on the basis that the code doesn't perform the beam centre finding routine if no direction has been selected. Typically we try to use the log levels you mention as follows:

- Notice - to provide very high-level informational messages when everything is OK.
- Warning - for when something isn't quite right but it's still possible to continue with the process.
- Error - when something isn't right such that we can't continue the process at all.

From the various comments, it sounds like if we improve the description of the message that's printed and prevent the crash that happens afterwards then that should resolve things here.

rbauststfc, 2024-09-27:
Yes @dehoni, you're right that you couldn't use radio buttons here with just two options. We could potentially change it to be a radio button with three options (`Left/Right`, `Up/Down` and `Both`), and that would prevent it being possible to make invalid selections.

However, on balance, I wouldn't recommend doing that at this stage. Looking back through the code I think this problem has been there for many years and this is the first time it's been noticed, so it doesn't currently seem to be having a lot of impact. Switching to radio buttons will be a slightly bigger change to implement (due to the changes to the internal logic that are required to support this) and will make the functionality slightly more complex to maintain. On that basis, to make best use of your available development resource, I'd recommend that we start with the lighter-touch solution here (update error message and tooltips and prevent the crash). If people start finding that they're coming across the error message more frequently then of course we could then look at switching to radio buttons. Hope that sounds OK, but just shout if not.

PR #38096 (2024-10-02): Fix ISIS SANS beam centre finder exception when no direction is selected
### Description of work

#### Summary of work

Prevents an unexpected exception when the beam centre finder is run with none of the Direction checkboxes selected. This fix changes the code to raise an error when validating the direction settings rather than simply returning nothing. This signals a failure to the asynchronous process, which prevents us incorrectly continuing down the success pathway of the code. Also ensures that the error message that's printed gives clearer instructions so that the user knows how to reso [... cut]
closed 2024-10-02 by github-actions[bot]