mantidproject/mantid #38143: Out of bounds error in IntegratePeaksShoeboxTOF

RichardWaiteSTFC, 2024-10-03:
**Describe the bug**

```
Error in execution of algorithm IntegratePeaksShoeboxTOF:
index 0 is out of bounds for axis 0 with size 0
  at line 434 in '/mnt/ceph/home/rw1061637/.conda/envs/mantid-unstable/plugins/python/algorithms/IntegratePeaksShoeboxTOF.py'

```
**To Reproduce**
Found integrating predicted peaks in run SXD34361 
(1) Ask me for .nxs files of workspace and peaks
(2)  Run this script

```
ws = Load(Filename='SXD34361_processed.nxs', OutputWorkspace='SXD34361')
peaks = Load(Filename='SXD34361_peaks.nxs', OutputWorkspace='SXD34361 _peaks')

params = {"NRows": 7, "NCols": 7, "GetNBinsFromBackToBackParams": True, 
          "NFWHM": 3, "NShoeboxInWindow": 3, 
          "WeakPeakThreshold": 20, 
          "WeakPeakStrategy": "NearestStrongPeak",
          "IntegrateIfOnEdge": False, "LorentzCorrection": True}
          
peaks_int = IntegratePeaksShoeboxTOF(InputWorkspace=ws, PeaksWorkspace=peaks, OutputWorkspace='peaks_int', **params)
```

**Expected behavior**
Would not crash - the problem seems to be here

https://github.com/mantidproject/mantid/blob/84edbc443c1b7a7b06cf69b2b5039e3551e58447/Framework/PythonInterface/plugins/algorithms/IntegratePeaksShoeboxTOF.py#L434

It looks like `weak_pk.ispec` is not in `ispecs` perhaps because the shoebox has been moved/optimised?

**Screenshots**

**Platform/Version (please complete the following information):**
 - OS: [e.g. Windows, RHEL 7, Ubuntu, macOS]
 - OS Version:
 - Mantid Version [e.g. 6.0.0]

**Additional context**

warunawickramasingha, 2024-12-03:
Hi @RichardWaiteSTFC, if you have access to the files required to recreate the issue, could you please attach? Thanks.

RichardWaiteSTFC, 2024-12-03:
This script reproduces the issue
```
ws = Load(Filename='SXD34361.nxs', OutputWorkspace='SXD34361')
peaks = FindSXPeaksConvolve(InputWorkspace='SXD34361', PeaksWorkspace='SXD34361_peaks', ThresholdIoverSigma=10, NRows=3, NCols=3, GetNBinsFromBackToBackParams=True, NFWHM=3, ThresholdVarianceOverMean=1.25)

params = {"NRows": 7, "NCols": 7, "GetNBinsFromBackToBackParams": True, 
          "NFWHM": 3, "NShoeboxInWindow": 3, 
          "WeakPeakThreshold": 20, 
          "WeakPeakStrategy": "NearestStrongPeak",
          "IntegrateIfOnEdge": False, "LorentzCorrection": True}
          
peaks_int = IntegratePeaksShoeboxTOF(InputWorkspace=ws, PeaksWorkspace=peaks, OutputWorkspace='peaks_int', **params)
```

PR #39124 (2025-04-09): Fixed an out of bounds error coming from IntegratePeaksShoeboxTOF alg…
### Description of work
Fixed an out-of-bounds error coming from `IntegratePeaksShoeboxTOF` algorithm when Weaker peaks are integrated by optimizing the shoebox having `WeakPeakStratergy` set to `NearestStrongPeak`.

Fixes #38143. 

### To test:
1. Run the below script. The peaks should be integrated successfully without any errors.
```
ws = Load(Filename='SXD34361.nxs', OutputWorkspace='SXD34361')
peaks = FindSXPeaksConvolve(InputWorkspace='SXD34361', PeaksWorkspace='SXD34361_peaks', ThresholdIoverSigma=10, NRows=3, NCols= [... cut]
closed 2025-04-09 by peterfpeterson