mantidproject/mantid #38225: MacOS Interface Smoke Tests  :facepalm:

MohamedAlmaki, 2024-10-18:
We must test both the .dmg installer package and install direct from conda.

* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n mantid_rc1 -c mantid/label/v6.11.0rc1 mantidworkbench --yes
mamba activate mantid_rc1
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :facepalm:

## ISIS only, if possible, so catalogue access can be tested

- [ ] MantidWorkbench opens without errors or warnings 
- [ ] Every option in `Interface` opens a GUI 
- [ ] Load some test data 
- [ ] Access Catalogue through algorithms (use Facilities account, same as for IDAaaS): [CatalogLogin](https://docs.mantidproject.org/algorithms/CatalogLogin-v1.html) and [CatalogGetDataFiles](https://docs.mantidproject.org/algorithms/CatalogGetDataFiles-v1.html)
- [ ] Saving/loading projects works 
 - [ ] Alter preferences in [File > Settings](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/01_save_settings.html#settings) and check they are obeyed

SilkeSchomann, 2024-10-21:
Garnet, SNAPRed and Shiver show the message that they are not available. This is as expected as they need to be installed manually first before they are available as an interface.

SilkeSchomann, 2024-10-21:
DNS Reduction in Direct throws an error message:

```
Traceback (most recent call last):
  File "/Users/silke.schomann/mambaforge/envs/mantid_smoke/lib/python3.10/site-packages/workbench/app/mainwindow.py", line 393, in <lambda>
    action.triggered.connect(lambda checked_py, script=script: self.launch_custom_python_gui(script))
  File "/Users/silke.schomann/mambaforge/envs/mantid_smoke/lib/python3.10/site-packages/workbench/app/mainwindow.py", line 353, in launch_custom_python_gui
    self.interface_executor.execute(open(filename).read(), filename)
  File "/Users/silke.schomann/mambaforge/envs/mantid_smoke/lib/python3.10/site-packages/mantidqt/widgets/codeeditor/execution.py", line 153, in execute
    executor.execute(code_str, filename, flags, self.globals_ns, line_offset)
  File "/Users/silke.schomann/mambaforge/envs/mantid_smoke/lib/python3.10/site-packages/mantidqtinterfaces/DNS_Reduction.py", line 24, in <module>
    from mantidqtinterfaces.dns_powder_tof.main_widget import DNSReductionGuiWidget
  File "/Users/silke.schomann/mambaforge/envs/mantid_smoke/lib/python3.10/site-packages/mantidqtinterfaces/dns_powder_tof/main_widget.py", line 13, in <module>
    from mantidqtinterfaces.dns_powder_tof.dns_modus import DNSModus
  File "/Users/silke.schomann/mambaforge/envs/mantid_smoke/lib/python3.10/site-packages/mantidqtinterfaces/dns_powder_tof/dns_modus.py", line 29, in <module>
    from mantidqtinterfaces.dns_single_crystal_elastic.options.elastic_single_crystal_options_widget import DNSElasticSCOptionsWidget
ModuleNotFoundError: No module named 'mantidqtinterfaces.dns_single_crystal_elastic'
```

RabiyaF, 2024-10-21:
I also tested this on the standalone version and saw the same Issue with DNS Reduction (which opened the Error Report dialog). And the same messages with Garnet, SNPRed and Shiver.

SilkeSchomann, 2024-10-21:
There is a fix now for DNS Reduction (https://github.com/mantidproject/mantid/pull/38252) so I am closing this issue.

closed 2024-10-21 by SilkeSchomann