mantidproject/mantid #38226: MacOS Plotting Smoke Tests  :chart_with_upwards_trend: :bar_chart:

MohamedAlmaki, 2024-10-18:
We must test both the .dmg installer package and install direct from conda.

* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n mantid_rc1 -c mantid/label/v6.11.0rc1 mantidworkbench --yes
mamba activate mantid_rc1
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :chart_with_upwards_trend: :bar_chart:

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
 - [ ] Simple plot 
 - [ ] Another way to plot 
 - [ ] Adding curves to existing plots 
 - [ ] Also, test out [waterfall](https://docs.mantidproject.org/nightly/plotting/WaterfallPlotsHelp.html#waterfall-plots) and [tiled]( https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#tiled-plots)
 - [ ] Check Toolbar buttons

* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
 - [ ] Plot all spectra 
 - [ ] Change colour map 
 - [ ] [Contour plot](https://docs.mantidproject.org/nightly/plotting/ColorfillPlotsHelp.html#contour-plots) (under 3D menu) 
 - [ ] Check Toolbar buttons

* [3D plotting](https://docs.mantidproject.org/nightly/plotting/3DPlotsHelp.html):
 - Load some data eg `LOQ74041` from the ISIS sample data
 - [ ] 3D surface
 - [ ] 3D wire frame  
 - [ ] Check Toolbar buttons

 ## Sliceviewer
 - [ ] Overly long instructions (don't spend  long!) and data [here](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html). In particular try editing the data in a workspace while it is open in Sliceviewer!

RabiyaF, 2024-10-21:
The only thing I was not able to test was the contour plots steps. That button did not show up on my UI. Everything else is working well on the Mac standalone version.

[image]

RabiyaF, 2024-10-21:
Ohh okay 👍🏼 I can change the colour of the contours. I was plotting it and then trying to find the paint palette icon which was the wrong way of doing it. But now everything works.

closed 2024-10-21 by SilkeSchomann