mantidproject/mantid #38231: Linux Fitting Smoke Tests  :muscle: :runner:

MohamedAlmaki, 2024-10-18:
* To install the tar.xz package for Linux, run `(sudo) tar -xJf mantid-VA.B.C.tar.xz` in a terminal and it will unzip the package in your current working directory. 
* To install via conda:
  - Use Intel Conda and make sure conda-forge is added to channels
```
mamba create -n mantid_rc1 -c mantid/label/v6.11.0rc1 mantidworkbench --yes
mamba activate mantid_rc1
workbench # to launch
```

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :muscle: :runner:

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [ ] Simple fit 
- [ ] Activating the fitting tools 
- [ ] Selecting the fitting range 
- [ ] Other Fit Function Settings properties 
- [ ] Setting up a fit model 
- [ ] Adjusting fit function parameter 
- [ ] Saving a model 
- [ ] Tying and constraining fit parameters 
- [ ] Execute your fit 
- [ ] Inspecting a fit result 
* Additional options [directions](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html)
- [ ] Non peak model + background 
- [ ] User defined function

MohamedAlmaki, 2024-10-21:
@GuiMacielPereira have you finished this test and are happy to close it?

GuiMacielPereira, 2024-10-21:
Yes, all is good, just had forgotten to comment