mantidproject/mantid #38240: IDAaaS Clean/Dirty Install Smoke Tests 🧼 💩

MohamedAlmaki, 2024-10-21:
An IDAaaS session link will be shared with you. It will either be a no-GPU or a GPU IDAaaS instance. All IDAaaS-installed packages are Conda installs.

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :soap: :hankey:

### Dirty install

* Make sure that you have several versions of Mantid installed
 * Last release
 * A nightly
 * If possible an old release
* Install the new release candidate of Mantid
- [x] Check that Mantid boots up correctly

### Clean install

* Remove all existing Mantid versions and associated files

**On Windows**:
 * Uninstall the program
 * Clear shortcuts from desktop
 * Remove the mantid  files in %APPDATA%

**On macOS** :
 * Remove the application
 * Remove the `~/.mantid directory`
 * Remove (or rename) `~/Library/Preferences/org.python.python.Python.plist`  and `~/Library/Preferences/org.python.python.plist` 

**On Linux** :
 * Remove the package: `/opt/Mantid`
 * Remove `~/.config/Mantid`
 * Remove `~/.mantid/`

Re-install the new release candidate of Mantid

- [x] Check that Mantid boots up correctly

rbauststfc, 2024-10-21:
I don't think the clean install test is relevant for IDAaaS, as multiple previous versions will always have been pre-installed? I tested installing into a Conda environment in a GPU workspace and all worked OK.

warunawickramasingha, 2024-10-21:
I've installed using the standalone in IDAaaS which is also not the usual case.

closed 2024-10-21 by rbauststfc