mantidproject/mantid #38241: IDAaaS Fitting Smoke Tests 💪 🏃

MohamedAlmaki, 2024-10-21:
An IDAaaS session link will be shared with you. It will either be a no-GPU or a GPU IDAaaS instance. All IDAaaS-installed packages are Conda installs.

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :muscle: :runner:

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [x] Simple fit 
- [x] Activating the fitting tools 
- [x] Selecting the fitting range 
- [x] Other Fit Function Settings properties 
- [x] Setting up a fit model 
- [x] Adjusting fit function parameter 
- [x] Saving a model 
- [x] Tying and constraining fit parameters 
- [x] Execute your fit 
- [x] Inspecting a fit result 
* Additional options [directions](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html)
- [x] Non peak model + background 
- [x] User defined function

rbauststfc, 2024-10-21:
@jhaigh0 or @warunawickramasingha, please would one of you be able to repeat the "Non peak model + background" test (https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html#non-peak-model-background)? I didn't get any errors but I couldn't reproduce the graphs that were shown. Could just be my lack of expertise at fitting!

Everything else I tested looked OK.

jhaigh0, 2024-10-21:
> @jhaigh0 or @warunawickramasingha, please would one of you be able to repeat the "Non peak model + background" test [...]

Yeah, for me the first part looks the same, but when the GausOsc is added nothing really changes. I think maybe some good starting parameters need to be added.

rbauststfc, 2024-10-21:
Discussing with others, we've confirmed that the problem is just getting the correcting starting parameters for the fit. We're going to open an issue to update the documentation and avoid confusion in future, but there's nothing else that needs to be done here.

closed 2024-10-21 by rbauststfc