mantidproject/mantid #38242: IDAaaS Interface Smoke Tests 🤦

MohamedAlmaki, 2024-10-21:
An IDAaaS session link will be shared with you. It will either be a no-GPU or a GPU IDAaaS instance. All IDAaaS-installed packages are Conda installs.

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :facepalm:

## ISIS only, if possible, so catalogue access can be tested

- [x] MantidWorkbench opens without errors or warnings 
- [x] Every option in `Interface` opens a GUI 
- [x] Load some test data 
- [x] Access Catalogue through algorithms (use Facilities account, same as for IDAaaS): [CatalogLogin](https://docs.mantidproject.org/algorithms/CatalogLogin-v1.html) and [CatalogGetDataFiles](https://docs.mantidproject.org/algorithms/CatalogGetDataFiles-v1.html)
- [x] Saving/loading projects works 
 - [x] Alter preferences in [File > Settings](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/useful/01_save_settings.html#settings) and check they are obeyed

jhaigh0, 2024-10-21:
When opening `Interfaces->Direct->DNS Reduction`
```
![image](https://github.com/user-attachments/assets/38cc222c-02b6-48fb-8bee-95ec10f9946e)
```

jhaigh0, 2024-10-21:
On using the `CatalogLogin` algorithm

[image]

SilkeSchomann, 2024-10-21:
There is a fix now for DNS Reduction (https://github.com/mantidproject/mantid/pull/38252) so I am closing this issue.

I am looking into the ICAT issue as it does not seem to be something we can influence. I will check on the ICAT slack channel, though.

closed 2024-10-21 by SilkeSchomann