mantidproject/mantid #38243: IDAaaS Plotting Smoke Tests 📈 📊

MohamedAlmaki, 2024-10-21:
An IDAaaS session link will be shared with you. It will either be a no-GPU or a GPU IDAaaS instance. All IDAaaS-installed packages are Conda installs.

**Before testing**:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

**Afterwards**:
- Comment below with any issues you came across.
- If no major issues were found, please **close this testing issue**.
- Check the main issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :chart_with_upwards_trend: :bar_chart:

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
 - [x] Simple plot 
 - [x] Another way to plot 
 - [x] Adding curves to existing plots 
 - [x] Also, test out [waterfall](https://docs.mantidproject.org/nightly/plotting/WaterfallPlotsHelp.html#waterfall-plots) and [tiled]( https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#tiled-plots)
 - [x] Check Toolbar buttons

* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
 - [x] Plot all spectra 
 - [x] Change colour map 
 - [x] [Contour plot](https://docs.mantidproject.org/nightly/plotting/ColorfillPlotsHelp.html#contour-plots) (under 3D menu) 
 - [x] Check Toolbar buttons

* [3D plotting](https://docs.mantidproject.org/nightly/plotting/3DPlotsHelp.html):
 - Load some data eg `LOQ74041` from the ISIS sample data
 - [x] 3D surface
 - [x] 3D wire frame  
 - [x] Check Toolbar buttons

 ## Sliceviewer
 - [x] Overly long instructions (don't spend  long!) and data [here](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html). In particular try editing the data in a workspace while it is open in Sliceviewer!

warunawickramasingha, 2024-10-21:
While following test instructions for [2D plots](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data) at step 3 in plot->Colorfill, select the scale of the X axis to be `Logit` and click ok/apply. There will be no data shown. Even changing the scale back to Linear/Log does not show any data unless the `Home` button is clicked.

warunawickramasingha, 2024-10-21:
The screenshot in Options menu in [here](https://docs.mantidproject.org/nightly/plotting/3DPlotsHelp.html) is outdated and needs updating docs.

warunawickramasingha, 2024-10-21:
> While following test instructions for [2D plots](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data) at [...]

This is **Not** a Regression

closed 2024-10-21 by SilkeSchomann