mantidproject/mantid #38254: Save single-crystal shoebox/skew integration on peak shape for subsequent overlap detection and two-step integration

RichardWaiteSTFC, 2024-10-21:
**Is your feature request related to a problem? Please describe.**
Q-space integration methods save the peak shape (sphere or ellipsoid) on the peak. For example `PeakShapeEllipsoid` here
https://github.com/mantidproject/mantid/blob/b7e0aaf70396afde9e7b1e83112f42e5372d2e13/Framework/DataObjects/src/PeakShapeEllipsoid.cpp#L16
This peak shape is currently used only for visualisation purposes.

**Describe the solution you'd like**
It would be good to save the detector IDs and TOF bin indices in a similar shape e.g.  `DetectorTOFShape` this could be used for:
- Overlap detection post integration (peaks could be excluded or flagged as combined)
- Two-step integration - re-integrate weak peaks using strong peak dimensions
- Eventual visualisation on instrument view (i.e. show bounding box of peak)

RichardWaiteSTFC, 2024-10-22:
Integration algorithms that require peak shape to be saved:
- `IntegratePeaksSkew`
- `IntegratePeaksShoebox`
- `IntegratePeaks1DProfile`

RichardWaiteSTFC, 2024-11-25:
Please leave `IntegratePeaks1DProfile` until after big refactor in #38439

warunawickramasingha, 2024-12-11:
Reopening the issue since the merged PR #38433  was meant to `partially fix` this issue and another PR linked to the issue is still in progress.

PR #38433 (2024-12-11): 38254 Introducing PeakShapeDetectorBin peak shape
### Description of work
A new peak shape `PeakShapeDetectorBin` was introduced to save the shape of each peak including the detector IDs and limits of either `TOF` or `dSpacing`. This shape can be used for

- Overlap detection post integration (peaks could be excluded or flagged as combined)
- Two-step integration - re-integrate weak peaks using strong peak dimensions
- Eventual visualisation on instrument view (i.e. show bounding box of peak)

Partially fixes #38254. 

#### Further detail of work

### To test:
1. Code review
2. Unit tests shou [... cut]
PR #38452 (2025-01-09): 38254 add PeakShapeDetectorbin to integration algorithms
### Description of work

#### Summary of work
The new peak shape `PeakShapeDetectorBin` is added to `IntegratePeaksShoeboxTOF` and `IntegratePeaksSkew` algorithms so that the detector ids and the limits of the integration are made available on the valid peaks integrated from those algorithms.

Fixes #38254. 

#### Further detail of work

### To test:

`IntegratePeaksShoeboxTOF` and `IntegratePeaksSkew` integration algorithms should run without making any errors and peaks of the the resulting peaks workspace should have the peak shape `Pe [... cut]
closed 2025-01-09 by robertapplin