mantidproject/mantid #38332: Plan to consolidate Nexus loading code

robertapplin, 2024-10-31:
**Describe the outcome that is desired.**
Mantid currently uses multiple mechanisms to load in data. This is outlined in the following issue https://github.com/mantidproject/roadmap/issues/21

These mechanisms primarily involve using the Nexus API library to load data, however this library is now unsupported. We therefore no longer want to use it, and it would be preferable to use the HDF5 API directly. This would provide us with several other benefits which were not previously available when using Nexus API, and could result in a performance improvement when loading data.

To do this switch, the following plan was discussed with @peterfpeterson and @martyngigg .

**Describe any solutions you are considering**
1. Pull all the Nexus API code into the `Framework/Nexus` location in Mantid
2. Remove XML, Fortran, etc. sections of this code
3. Move the two Muon HDF4 algorithms (`LoadMuonNexus1` and `LoadMuonNexus2`) to the `Framework/Muon` library to isolate them
4. Use HDF4 directly in these algorithms rather than using the Nexus API. (Link HDF4 to the Muon library, and add HDF4 to the conda environment)
5. Remove HDF4 sections of the Nexus API code within the `Framework/Nexus` location
6. Review the old wrappers such as `NexusFileIO` and `NexusIOHelper` to understand how they are used, and where they are used
7. Refactor these wrappers to simplify and make them "lightweight". Ideally we want to use one common mechanism to load HDF5 data across the Mantid codebase, and this mechanism might be a very lightweight wrapper around the HDF5 library

peterfpeterson, 2024-10-31:
As of the time of this writing, the dependencies of the current nexus conda package are (using [conda-tree](https://github.com/conda-incubator/conda-tree))
```sh
$ conda tree deptree
...
  ├─ nexus 4.4.3 [required: 4.4.*]
  │  ├─ hdf4 4.2.15 [required: >=4.2.15,<4.2.16.0a0]
  │  │  ├─ libgcc-ng 14.1.0 [required: >=12]
  │  │  │  └─ dependencies of libgcc-ng displayed above
  │  │  ├─ libjpeg-turbo 3.0.0 [required: >=3.0.0,<4.0a0]
  │  │  │  └─ dependencies of libjpeg-turbo displayed above
  │  │  ├─ libstdcxx-ng 14.1.0 [required: >=12]
  │  │  │  └─ dependencies of libstdcxx-ng displayed above
  │  │  └─ libzlib 1.3.1 [required: >=1.2.13,<2.0.0a0]
  │  │     └─ dependencies of libzlib displayed above
  │  ├─ hdf5 1.14.3 [required: >=1.14.3,<1.14.4.0a0]
  │  │  └─ dependencies of hdf5 displayed above
  │  ├─ libgcc-ng 14.1.0 [required: >=12]
  │  │  └─ dependencies of libgcc-ng displayed above
  │  ├─ libjpeg-turbo 3.0.0 [required: >=3.0.0,<4.0a0]
  │  │  └─ dependencies of libjpeg-turbo displayed above
  │  ├─ libstdcxx-ng 14.1.0 [required: >=12]
  │  │  └─ dependencies of libstdcxx-ng displayed above
  │  ├─ libxml2 2.12.7 [required: >=2.12.6,<3.0a0]
  │  │  └─ dependencies of libxml2 displayed above
  │  ├─ libzlib 1.3.1 [required: >=1.2.13,<2.0.0a0]
  │  │  └─ dependencies of libzlib displayed above
  │  ├─ ncurses 6.5 [required: >=6.4.20240210,<7.0a0]
  │  │  └─ dependencies of ncurses displayed above
  │  └─ readline 8.2 [required: >=8.2,<9.0a0]
  │     └─ dependencies of readline displayed above
```

martyngigg, 2024-11-01:
>     6. Move code from napi5.c into napi.c to remove a layer of abstraction. This may be better to combine with the next step.
> 
>     7. Move the C code of napi.c into NeXus::File.cpp
> 
>     8. Review the old wrappers such as `NexusFileIO` and `NexusIOHelper` to understand how they are used, and where they are used
> 
>     9. Refactor these wrappers to simplify and make them "lightweight". Ideally, we want to use one common mechanism to load/save HDF5 data across the Mantid codebase, and this mechanism might be a very lightweight wrapper around the HDF5 library

I think this looks good and is a good summary of our discussion 😸. I'd probably say with steps 6-7 might be better done after 8 and the review step encompasses what an ideal API for the new internal `NeXusFile` class would look like and how it interacts with things like NexusIOHelper rather than just picking up the neXus API code wholesale and putting it inside.

I also would say that I think `NexusFileIO` should be deleted as it's pretty ancient and whatever the new `NeXus::File/NexusIOHelpers` look like can probably subsume all of that behaviour in a more modern way.

peterfpeterson, 2024-11-08:
We also need to figure out how https://github.com/mantidproject/mantid/blob/main/Framework/DataHandling/inc/MantidDataHandling/H5Util.h fits into things

peterfpeterson, 2024-11-11:
Consider not [setting the filecache size](https://github.com/nexusformat/code/blob/5b803b3a0014bd9759b3d846da3cd3c1cfafd7d5/src/napi5.c#L280-L309) automatically during file opening. This was suggested by @ekapadi.

robertapplin, 2024-12-10:
List of load algorithms which use the `NexusDescriptor` class:
- LoadEMU
- LoadILLDiffraction
- LoadILLIndirect2
- LoadILLLagrange
- LoadILLPolarizedDiffraction
- LoadILLReflectometry
- LoadILLSALSA
- LoadILLSANS
- LoadILLTOF2
- LoadISISNexus2
- LoadLLB
- LoadMLZ
- LoadMuonNexus1 & 2
- LoadNXSPE
- LoadNXcanSAS
- LoadPLN
- LoadQKK
- LoadSINQFocus
- LoadSassena
- LoadTOFRawNexus
- LoadGeometry
- UpdateInstrumentFromFile

peterfpeterson, 2024-12-13:
For reference: [H50 documentation](https://portal.hdfgroup.org/documentation/hdf5/latest/group___h5_o.html#ga0090da86c086c1c63a5acfaed39a035e) in HDF5

peterfpeterson, 2025-01-15:
After doing a lot of searching and whatnot using [ripgrep](https://github.com/BurntSushi/ripgrep) to find what includes what (see attached svg), I have discovered that there are 7 was mantid uses for accessing "nexus" files. Things that are external are marked that way. Some methods are intermingled.
* `napi.h`
  * low level c-based API that should be gotten rid of
  * used by `LoadILLDiffraction`, `NexusClasses` (mentioned below), `LoadHelper` (mentioned below)
* `hdf5.h` (from HDF)
  * low level c-based API that should be gotten rid of
  * used by `LoadMDTest`, `LoadSassena`, `SaveNXSPETest`, `NexusHDF5Descriptor` (should be merged with `NexusDescriptor`
* `H5Cpp` (from HDF) and `H5Util`
  * higher level cpp-based API
  * used by 27 files
  * plan to keep
* `NexusFile` + `NexusIOHelper`
  * higher level cpp-based API
  * used by 52 files
  * plan to keep, maybe combine
  * underlying implementation should be replaced with `H5Cpp` / `H5Util`
* `NexusClasses`
  * written using `napi.h`
  * abstraction very similar to `H5Cpp`, but with some nexus specifics
  * used by `LoadMuonNexus1` (hdf4), `LoadMuonNexus2` (hdf4), `LoadMuonNexusV2` (hdf5), `LoadNexusProcessed` (replaced with a v2 that uses `H5Cpp` directly), and `LoadSINQFocus`
  * may be possible to replace its usage in `LoadNexusProcessed` and `LoadSINQFocus` to use `H5Cpp` equivalents
  * existing version should be moved/copied into `Framework/Muon/` where it can be refactored to only work with hdf4
* `NexusFileIO`
  * written using `NexusFile`
  * used exclusively by `SaveNexusProcessed`
  * should be moved to `FrameworkDataHandling` and see what of it can be refactored away. Possibly rename the file to make it clearer that it is only used by `SaveNexusProcessed`
* `LoadHelper`
  * written using `napi.h`
  * used by `LoadEventNexus`, `LoadILLDiffraction`, `LoadILLIndirect2`, `LoadILLLagrange`, `LoadILLPolarizedDiffraction`, `LoadILLReflectometry`, `LoadILLSALSA`, `LoadILLSANS`, `LoadILLTOF2`, `LoadLLB`, `LoadMLZ`, `LoadSINQFocus`, `LoadSwans`
  * should find equivalents in `NexusFile` and `NexusIOHelper` and transition codes to use those instead

Based on pervasiveness and completeness, my preference is to move things to use `H5Cpp`/`H5Util` or `NexusFile`/`NexusIOHelper`. These have fairly different access models (analogous to DOM and SAX). I will be reworking the description above to reflect this new understanding.

peterfpeterson, 2025-08-14:
All the work has been done

closed 2025-08-14 by peterfpeterson