mantidproject/mantid #38394: Save on Elwin does not save in expected directory

robertapplin, 2024-11-13:
**Reported by:** Spencer Howells

**Description**
Saving the result of running the Elwin tab does not save the workspace in the expected directory. We should expect that the save button will save the workspace in the default save directory.

**To Replicate**
1. Set the default save directory in the Manage User Directories widget
2. Open Inelastic->Data Processor interface
1. Load data into Elwin
2. Click Run
3. Click `Save` in the bottom left of the Elwin tab
4. The result data is not saved in the default save directory

robertapplin, 2024-12-03:
I can't seem to replicate this problem. @SpencerHowells are you still able to replicate this?

SpencerHowells, 2024-12-03:
@robertapplin Seems to be Ok on nightly. Probably me not changing directory.

robertapplin, 2024-12-03:
Closing this because we could not replicate