mantidproject/mantid #38449: Add Move to the Top button on Manage User Directories interface

adriazalvarez, 2024-11-26:
**Is your feature request related to a problem? Please describe.**

A user has reported through the Mantid help mail a cumbersome workflow when handling multiple search paths in the Data Search Directories list: 
_"There is a significant penalty in Mantid to having a very large number of search paths in the manage user directories in Mantid on IDAaaS. Given that mantid paths are autoconfigured on IDAaaS adding new folders is quite an irritating process.
Can the addition of a “Move to top” button be considered in addition to the “move up” and “move down” ones that exist on the interface"_

**Describe the solution you'd like**

Adding a button on the Manager User Directories interface which sends the selected search directory to the top. 
[image]
**Describe alternatives you've considered**

**Additional context**

adriazalvarez, 2024-12-10:
If there are a large number of search paths, as Robert seems to be using, dragging the item up may be cumbersome too. I would personally prefer a button.

closed 2024-12-16 by SilkeSchomann

jhaigh0, 2024-12-16:
Be sure to let Rob know this has been done for him.

PR #38497 (2024-12-16): Added a move to top button in the manage user directory UI
### Description of work

#### Summary of work

Fixes #38449 . 

#### Further detail of work

### To test: