mantidproject/mantid #38457: Get rid of MantidKernel/System.h

peterfpeterson, 2024-11-27:
**Describe the outcome that is desired.**

Remove [`System.h`](https://github.com/mantidproject/mantid/blob/v6.11.0/Framework/Kernel/inc/MantidKernel/System.h) since it is created in the generate files called `DllConfig.h` by way of [`GenerateMantidExporteHeader.cmake`](https://github.com/mantidproject/mantid/blob/v6.11.0/buildconfig/CMake/GenerateMantidExportHeader.cmake)

**Describe any solutions you are considering**

This can be done in stages, but each removal from `System.h` will invalidate almost the entire build tree. Consider that when planning out the work.

* Move the `#pragma warning(disable:` commands from L33, 37, and 42 to exist as compiler flags in [`WindowsSetup.cmake`](https://github.com/mantidproject/mantid/blob/v6.11.0/buildconfig/CMake/WindowsSetup.cmake). When doing the move, consider whether the flags are still necessary.
* **DO SECOND** Remove direct `#include "MantidKernel/System.h"` from everything. Some sub-packages predate the existence of `DllConfig.h` being created and should use that instead. There are currently 503 files that directly include `MantidKernel/System.h`.
* The macros for exporting symbols on L34-55 is generated already by `GenerateMantidExportHeader.cmake` and can be removed
* The macros for deprecating symbols on L70-77 is generated already by `GenerateMantidExportHeader.cmake` and can be removed
* The macro for UNUSED_ARG on L63-65 is generated already by `GenerateMantidExportHeader.cmake` and can be removed. Add a comment/link in the generated `DllConfig.h` that one should consider [`maybe_unused`](https://en.cppreference.com/w/cpp/language/attributes/maybe_unused) instead.
* Remove the includes for `cstddef` and `cstdint`. If this generates an excess of changes, move the includes into `GENERATE_EXTERN` clause of [`GenerateMantidExportHeader.cmake`](https://github.com/mantidproject/mantid/blob/v6.11.0/buildconfig/CMake/GenerateMantidExportHeader.cmake)
* **DO LAST** Remove `MantidKernel/System.h` and update [`GenerateMantidExportHeader.cmake`](https://github.com/mantidproject/mantid/blob/v6.11.0/buildconfig/CMake/GenerateMantidExportHeader.cmake) to not include it.

**NOTE ABOUT `UNUSED_ARG`**

This is the main reason why tests include `MantidKernel/System.h`. Either change them to use the sub-package specific `DllConfig.h` or change to [`maybe_unused`](https://en.cppreference.com/w/cpp/language/attributes/maybe_unused)

peterfpeterson, 2024-12-18:
From a small test, removing `cstddef` from `MantidKernel/System.h` can be done without much work.

martyngigg, 2025-01-16:
Just a note.

The Windows `#pragma warning` in `System.h` should be able to move to compiler flags in [WindowsSetup](https://github.com/mantidproject/mantid/blob/b40ff3d43c0ebc13e0f580c5969ef604e4026734/buildconfig/CMake/WindowsSetup.cmake#L30) such as `/wd4251`. https://learn.microsoft.com/en-us/cpp/build/reference/compiler-option-warning-level?view=msvc-170 for reference.